shaise/FreeCAD_SheetMetal #1: Allow to use negative gap values for extrude objects

deveee, 2015-09-02:
It is useful in some cases and it seems to be not a problem to allow it.

I'm not creating a pull request because I'm not sure if it doesn't have any side effects. Here is the patch:

```
diff --git a/SheetMetalCmd.py b/SheetMetalCmd.py
index ce89972..1671799 100644
--- a/SheetMetalCmd.py
+++ b/SheetMetalCmd.py
@@ -40,7 +40,7 @@ def smMakeFace(edge, dir, from_p, to_p):
     return Part.Face(w)
 
 
-def smBend(bendR = 1.0, bendA = 90.0, flipped = False, extLen = 10.0, gap1 = 0.0, gap2 = 0, reliefW = 0.5, 
+def smBend(bendR = 1.0, bendA = 90.0, flipped = False, extLen = 10.0, gap1 = 0.0, gap2 = 0.0, reliefW = 0.5, 
             reliefD = 1.0, selFaceNames = '', MainObject = None):
             
   #AAD = FreeCAD.ActiveDocument
@@ -256,14 +256,14 @@ class SMExtrudeWall:
     selobj = Gui.Selection.getSelectionEx()[0]
     
     obj.addProperty("App::PropertyLength","length","Parameters","Length of wall").length = 10.0
-    obj.addProperty("App::PropertyLength","gap1","Parameters","Gap from left side").gap1 = 0.0
-    obj.addProperty("App::PropertyLength","gap2","Parameters","Gap from right side").gap2 = 0.0
+    obj.addProperty("App::PropertyFloat","gap1","Parameters","Gap from left side").gap1 = 0.0
+    obj.addProperty("App::PropertyFloat","gap2","Parameters","Gap from right side").gap2 = 0.0
     obj.addProperty("App::PropertyLinkSub", "baseObject", "Parameters", "Base object").baseObject = (selobj.Object, selobj.SubElementNames)
     obj.Proxy = self
 
   def execute(self, fp):
     #s = smExtrude(extLength = fp.length.Value, selFaceNames = self.selFaceNames, selObjectName = self.selObjectName)
-    s = smBend(bendA = 0.0,  extLen = fp.length.Value, gap1 = fp.gap1.Value, gap2 = fp.gap2.Value, reliefW = 0.0,
+    s = smBend(bendA = 0.0,  extLen = fp.length.Value, gap1 = fp.gap1, gap2 = fp.gap2, reliefW = 0.0,
                 selFaceNames = fp.baseObject[1], MainObject = fp.baseObject[0])
     fp.Shape = s
     

```

shaise, 2015-09-07:
Negative gaps creates illegal sheet metal parts. Can you give example of use?

deveee, 2015-09-07:
I mean more or less something like this:
[image]

It seems to be not possible to create it without negative gaps.

shaise, 2015-09-07:
Can you send the model?

shaise, 2015-09-07:
OK. Got it now.
Negative gaps on extruded objects are now implemented.

closed 2015-09-07 by shaise

deveee, 2015-09-08:
Thanks a lot!