shaise/FreeCAD_SheetMetal #2: 180 degrees cylinder sometimes is flattened in wrong direction

deveee, 2015-09-02:
Bend angle has wrong value. In logs for my two faces I saw:
```
Face61 Type: Bend bend_dir: up
Face_idx 60 bend_angle -180.0
Face61 k-factor: 0.65
angle: -180.0 test_angle: 45.0 trans: 4.16261026601
(...)
Face17 Type: Bend bend_dir: up
Face_idx 16 bend_angle -180.0
Face17 k-factor: 0.65
angle: -180.0 test_angle: 45.0 trans: 4.16261026601
```
In this case unfolding has been done wrongly.

But when I saved the file, closed FreeCAD, and then reopened it, I got proper bend angle value:
```
Face61 Type: Bend bend_dir: up
Face_idx 60 bend_angle -180.0
Face61 k-factor: 0.65
angle: -180.0 test_angle: 45.0 trans: 4.16261026601
Face17 Type: Bend bend_dir: up
(...)
Face_idx 16 bend_angle 180.0
Face17 k-factor: 0.65
angle: 180.0 test_angle: 45.0 trans: 4.16261026601
```
Second angle is non-negative, and unfolding started working properly.

I suppose that it is a problem with rounding or different precision somewhere... Rather hard to determine what is the reason.

shaise, 2015-09-07:
Can you tell me how to reproduce this problem?

deveee, 2015-09-07:
You can just create something like this:
[image]

I don't know what I did previously that it started working properly. Because now reopening document doesn't solve this.

On linux I have freecad 0.14, on windows version 0.15. And I can reproduce it on both.

Not sure if it matters, but thickness is 1 mm and every angle is 180° and 2 mm. Everything other has default values.

shaise, 2015-09-07:
Ok, managed to reproduce the problem. It has nothing to do with negative values. Try 181 degrees, and you will see the unfold is correct even though it logs both 181 and -181 bends.
I will try to find the problem.

shaise, 2016-07-14:
The latest version of the workbench, using Ver 17 of Ulrich's unfolder, seems to solve this issue

closed 2016-07-14 by shaise

deveee, 2016-07-15:
It works fine now. Thanks a lot!