shaise/FreeCAD_SheetMetal #4: Unfolding Problem when two Flanges meet at Right Angles and the Bend Radius equals the Pad Length

DoctorBit, 2016-04-21:
To reproduce the problem:
1) Create a sketch of a right-angled triangle.
2) Pad to 1 mm (not the default of 10 mm). This matches the default bend radius of 1 mm.
3) Add bends to both legs of the right-angled triangle (not the hypotenuse!).
4) Unfold

Notes: 
This bug only happens when:
1) The sketch contains a right angle - an equilateral triangle unfolds without problem.
2) The bend radius exactly matches the pad thickness.

ulrich1a, 2016-04-25:
You may know that Shaise had integrated my macro for sheet unfolding into the workbench.
You may try this new version of the macro.
Unfortunately I have actual no time to make the changes for including it into the workbench.
[sheet_ufo17.py.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/235406/sheet_ufo17.py.zip)

Ulrich

shaise, 2016-04-29:
The latest unfolder works on this example. Thank you Ulrich.
I have updated the WB to use this latest version.
DoctorBit,
Please test and comment.
Shai

DoctorBit, 2016-04-29:
Thanks guys, that got it!  I'm marking this issue closed. :+1:

closed 2016-04-29 by DoctorBit