shaise/FreeCAD_SheetMetal #5: unfolding fails.

sliptonic, 2016-06-21:
I have a relatively simple model that fails unfolding.  I'm attaching the model and the error log
Just remove the .zip extension from the project file.

[unfoldfaillog.txt](https://github.com/shaise/FreeCAD_SheetMetal/files/326498/unfoldfaillog.txt)
[foldingbug.fcstd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/326501/foldingbug.fcstd.zip)

shaise, 2016-07-14:
Hi sliptonic,
Sorry for the late response.
This issue happens when you add folds to an already unfolded sheet, then try to unfold again.
I have fixed this issue, please check.
Do not check it on the file you sent as the unfolded object there is created incorrectly. You will need to redo your work from the initial folded object.
Let me know if it works and close the issue.

luzpaz, 2017-07-18:
@sliptonic ping

luzpaz, 2018-09-15:
@sliptonic hard ping :stuck_out_tongue_winking_eye:

sliptonic, 2019-07-09:
Can't duplicate problem anymore.  Please consider closed.  Thank you and sorry for late response.

closed 2019-07-09 by sliptonic