shaise/FreeCAD_SheetMetal #6: Another unfolding failed

deveee, 2016-10-28:
Unfolding fails with error:
Error Analysis: counter face not found at Face37

It occurs with current plugin version with freecad 0.16.

It seems to be related to the rectangle on Sketch002. When I remove it or use some particular dimensions, it works fine.

Here is the file and log from console:
[unfold_log.txt](https://github.com/shaise/FreeCAD_SheetMetal/files/558342/unfold_log.txt)
[emi-prototyp7-ramka-ekran_10_1.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/558343/emi-prototyp7-ramka-ekran_10_1.zip)

If you need, I can test some modifications, provide some additional debug info etc.

deveee, 2016-11-07:
Here is another example. It looks like a problem with some tolerances because it doesn't work with bend length 0,4 mm and it works with length 5 mm. But original object is much more complicated and this 0,4 mm is needed there ;)

[test.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/574738/test.zip)

luzpaz, 2017-07-18:
@deveee is this still an issue with the latest 0.17 FC and latest revision of SheetMetal ?

luzpaz, 2019-07-11:
@shaise probably close this ticket as well.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise