shaise/FreeCAD_SheetMetal #8: sheetmetal loosing base face when changing gaps of extrude

enkidu, 2016-12-12:
how to reproduce:
1. create pad from sketch, wake a tray with foldings and extrudes;
2. map sketch to any face of the pad, make cutout in pad design
3. change any gap parameter of initial extrudes - next one will remap to improper face

It can be however FreeCAD engine limitation - but I am not a python enthusiast

shaise, 2016-12-12:
This is indeed a FreeCad limitation. Whenever you construct an object, and then try to change the base object, it might break. The reason is that FreeCad names objects like faces Face1, Face2 etc.. When you map a sketch, it maps to, say Face3. But when you change the base object by changing gaps, this might change the names of the faces, so Face3 is now on a different face.

closed 2016-12-12 by shaise

deveee, 2016-12-12:
Is there a way to workaround it?

shaise, 2016-12-12:
I wish there was a way. This issue is problematic in any parametric cad design. It is true not only for sheet metal. Many times when I made changes to a base object, all my fillets went bad as they switched to other edges. In most of the cases I found out that if the changes to the base object does not change the amount of faces created by this operation, then everything will be OK. otherwise it will mess with your end result.

shaise, 2016-12-12:
Perhaps you might ask around on freecad forum, maybe they plan a solution for that.
http://forum.freecadweb.org/