shaise/FreeCAD_SheetMetal #9: 0.17 Support

Tarcontar, 2017-02-15:
Hi,
is there a chance this tool gets updated for version 0.17?
Since I cant realy create sketches on faces created with this tool anymore in 0.17, but need this version for compatibility with another tool(dimensioning).

greetings Tarcontar

shaise, 2017-02-17:
Hi Tarcontar,

Yes, I can see its an issue with ver 0.17. I will try to solve it.
Meanwhile as a workaround, you can select the sheetmetal part, convert it
to body (by pressing the new body icon from part design workbench) and then
you can create sketches on faces as usual.
Let me know if this works.

shai.

Tarcontar, 2017-03-02:
Hi shai,

sorry for my late reply. The workaround you mentioned seems to work, but it would be nice if you could fix the issue with the current version.

Tarcontar

Von: Shai Seger
Gesendet: Freitag, 17. Februar 2017 22:34
An: shaise/FreeCAD_SheetMetal
Cc: Tarcontar; Author
Betreff: Re: [shaise/FreeCAD_SheetMetal] 0.17 Support (#9)

Hi Tarcontar,

Yes, I can see its an issue with ver 0.17. I will try to solve it.
Meanwhile as a workaround, you can select the sheetmetal part, convert it
to body (by pressing the new body icon from part design workbench) and then
you can create sketches on faces as usual.
Let me know if this works.

shai.

shaise, 2017-03-02:
Hi Tarcontar,

I did some more research into the new part design of freecad. It turns out that they have completely changed the "part design" workflow (for good reasons) but in this way making almost everything else incompatible. Even if you switch to the built in "Part" work bench and create, say a Cube, it will not be compatible with part design.
I am trying now to see if it is even possible for external workbenches to be compatible with part design. It is quite complicated, and for now, I cant promise I will be able to do it.

shai

shaise, 2017-03-04:
Hi Tarcontar,

Actually, I made some advances. If you take the V0.17 Branch, you will see that using SheetMetal on ne PartDesign objects now works. Just make sure the Body you work on is active.

shai

shaise, 2017-03-04:
OK, 
SheetMetal WB is now compatible with V0.17. But it required a change to main FreeCAD, so it can be used only with the next builds of FreeCAD (after today)

shaise, 2017-03-05:
Now works with the latest pre-build: 10423

closed 2017-03-05 by shaise