shaise/FreeCAD_SheetMetal #11: Unfold issue

matthewdickinson, 2017-03-23:
I was trying to unfold this part, but got this error when trying to unfold from the small flat face at the top of the part: "Error Analysis: did not find startangle of bend, please post failing sample for analysis at Face36"

Here are the part and log files from doing this unfold.
[B5.cleaned.STEP.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/865591/B5.cleaned.STEP.zip)
[postable-log.txt](https://github.com/shaise/FreeCAD_SheetMetal/files/865594/postable-log.txt)

Note: I have [this](https://github.com/shaise/FreeCAD_SheetMetal/pull/10) set of changes in my plugin code to allow me to get the actual error response rather than just a crash.

shaise, 2017-03-23:
The unfold script is not my code but belongs to ulrich1a. I have no direct
contact with him.
I have tried to debug it myself but could not fix it.
Ulrich sometimes follows freecad forum, so you might try there.

luzpaz, 2017-07-18:
CC @ulrich1a

imtaurus94, 2017-10-04:
Just by looking at your 3D model, and the logs, Looks like the error comes from Face12 and or Face15. They both have more than 4 edges. The script doesn't seem to be able to handle these very well.

luzpaz, 2017-10-28:
These issues should be posted to the FreeCAD forum to get potentially fixed

luzpaz, 2019-03-15:
@jaisekjames care to weigh in ?

shaise, 2019-05-20:
I have just tried this model, and it unfolds correctly no mater what face i select.

luzpaz, 2019-07-09:
@shaise should close this ticket then

closed 2019-07-09 by shaise