shaise/FreeCAD_SheetMetal #12: add animated gif to README

luzpaz, 2017-07-19:
would love to see an example of this WBs capabilities quickly. Please add something like this. Thanks!

shaise, 2017-07-19:
What do you mean by animated gif?

luzpaz, 2017-07-19:
a gif that has many gifs embedded that play in sequence to simulate a video.   
For example:   
[image]

shaise, 2017-07-19:
Can you suggest a software to generate it?

easyw, 2017-07-19:
https://getsharex.com/ for windows
https://github.com/phw/peek linux
http://www.omgubuntu.co.uk/2016/12/gifine-animated-gif-recorder-linux linux
https://itsfoss.com/best-gif-recorder-linux/

shaise, 2017-07-19:
easyw, Thanks for the great tools!
luzpaz, Thanks, done.

closed 2017-07-19 by shaise