shaise/FreeCAD_SheetMetal #17: unfold wok incorrect with simple model

yaroslavVl, 2017-07-28:
FreeCAD_0.17.11659_x64_dev_win
Latest SheetMetal
fig1 - ok:
[image]
fig2 - failed:
[image]
fig3,4 - strange shape:
[image]
[image]

[2.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1183591/2.zip)

imtaurus94, 2017-10-04:
The issue comes from the model. Face6 and Face12 are touching. Even in real life this wouldn't work. The bend script should be corrected to avoid making this possible.

luzpaz, 2017-10-28:
These issues should be posted to the FreeCAD forum to get potentially fixed

luzpaz, 2019-07-12:
@shaise can you weigh in on this ?

shaise, 2019-07-12:
by using the gap parameter of the bend, this issue can be avoided. I'm not sure though that doing that automatically is an easy task.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise