shaise/FreeCAD_SheetMetal #18: Unfold not working (invalid thickness

pasiz, 2017-08-22:
Every folded part, unfold gives "invalid thickness" error, worked yesterday.

pasiz, 2017-08-23:
Replicated with 0.16, unfold gives invalid thickness

pasiz, 2017-08-23:
estimated thickness: 4.27377890343 measured thickness: 8.54755780685
Error Starting: invalid thickness at Face1

No matter if i use 0.16 or 0.17 daily

imtaurus94, 2017-10-04:
Please add a 3D model to help debug this script

shrines, 2018-01-02:
I just installed the latest available sheetmetal master.  Invalid Starting Thickness still occurs when attempting to unfold in either FreeCAD 0.16 or 0.17.  A 3D model is attached that demonstrates the problem.
[unfold invalid thickness test.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1597932/unfold.invalid.thickness.test.zip)

shaise, 2018-01-02:
For me it works great with the attached sample.
Are you sure you selected the correct face before pressing unfold? you should select one of the big surfaces, the ones that will become the face of the unfolded sheet.

shrines, 2018-01-03:
Hi Shai,
It works great for me, too, when I select the proper face.  Thanks for the
support.
--Steve

shrines, 2018-01-05:
Hey Shai,

As a follow-up to my previous post I have attached a simple sheet metal
drawing that won't unfold.

I created this example using FreeCAD 0.16.6706 running on Windows 7 (My
company is a bit behind updating OS).

Regards,
--Steve

shaise, 2018-01-06:
missing attachment?

shrines, 2018-01-07:
Hi Shai,

Okay, I started from scratch to recreate the problem using FC 0.17.12871 on
Windows 10 with the latest Sheet Metal Workbench installed.

I created a base rectangle 1.5 x 2.5" using the Part Design Workbench and
padded it to 0.1".  I then switched to the Sheet Metal Workbench and bent
all four edges (bend, bend1, bend2 and bend3) with the default radius and
length.  All is good so far.  I then defined a new sketch in Part Design,
defined two holes in bend3 and created a pocket.  All is still OK.

At that point I went back to bend2 and changed the value of the gap2
parameter  from 0.0 to 1.0 (which should not have had any effect on bend3).

After that edit the axis of bend3 was rotated 90 degrees on the Y axis
which removed the holes from bend3 and broke the part.

As long as the part is created in sequence in the Sheet Metal Workbench
everything works fine.  But, as soon as an edit is made at an earlier step
in the parametric sequence, then any bends created after the edit have
their drawing plane reference screwed up.  The working planes of subsequent
sketches are also rotated 90 degrees.  (In FC 0.16.6706 this has the effect
of breaking established relationships between bend faces and sketches.)

The attached sample is the end result of following the above steps.

Kind regards,
--Steve

shaise, 2018-01-07:
Hi Steve,

What you describe here, is a fundamental flaw in freeCad's cad engine. This is not only a problem of sheet metal. When any part is created, freecad assign numbers to each face (face1, face2 etc..) when you are working with part design and attach a sketch to a face (such the one of bend3) freecad actually attach it to say "face24". Same for bends - if you start a bend from face5, this is what freecad will remember - that the bend is attached to face5. Now when you go back and change anything up the tree, as long as it does not change the total number of faces (say: change slightly thickness of pad, or make one bend longer), this will work ok. However, when adding a "gap" to bend 2, it will create a step that creates a new part with more faces. So now after the new numbering, the same face selected for bend3 is now  on a totally different place and thus breaking the part.

The guys on freecad are working on a better naming for faces to reduce this sort of issues, but they are still long way from that.

shrines, 2018-01-07:
Shia,

Thank you for the clear explanation.  I really appreciate your help.

--Steve

luzpaz, 2018-09-15:
@shaise can this ticket be closed ?