shaise/FreeCAD_SheetMetal #20: Focus at completion of bend

shrines, 2018-01-13:
Hi Shai,

If I make a bend and then make a second bend,  in the Tree View the second bend will be shown in bold and the previous bend will be shown with a gray highlight.  The information shown in the Property View reflects the information from the first bend.

Is this intended behavior?  

I ask because I frequently screw up my model by creating a new bend and immediately adjusting the bend properties --only to find that I have inadvertently modified the properties of the previous bend, not the bend I just created.

I think it would improve bend work flow if, as the last step in a bend operation, you set the focus to the new bend rather than leaving it on the previous bend.

Thanks,
--Steve

shaise, 2018-01-13:
I agree. I will look into it as soon as i have time

luzpaz, 2019-07-17:
Please tag as 'enhancement'

luzpaz, 2020-03-01:
@shaise please tag as 'enhancement'  thanks

luzpaz, 2021-04-06:
@shaise is this on the roadmap ?

shaise, 2021-04-08:
Hi @luzpaz ,

the wb now deselects everything after creating a sheetmetal feature, as every other part design feature. So there is no way to make that mistake now. You now have to specifically select the feature if you want to make changes. So I consider this issue as fixed.

shai

luzpaz, 2021-04-08:
> wb now deselects everything after creating a sheetmetal feature

This is separate from the OPs FR, but would still make sense that the Sheetmetal wb also have a solver like functionality that tries to intuit where the next bend will be requested (or highlight different junctions as possibilities?)

shaise, 2021-04-08:
IMHO it is impossible to guess the next SheetMetal operation. Any workflow is good, and some non SM features can be added along the way.