shaise/FreeCAD_SheetMetal #21: No response on unfold

shrines, 2018-01-23:
Hi Shai,

I have stumbled on another odd behavior in the sheet metal workbench.  

I created the attached model using FC 0.17.13005.  (Remove .zip in the filename to open.)

The rear chassis and the bottom left cover parts unfold as expected.  

The bottom right cover fails to unfold creating several shapes in the process.  --This acts like the part was broken due to new faces being added by editing, so I will rebuild that part and not worry about it.

The reason I am writing is that, when I attempt to unfold either the top left cover or the top right cover nothing happens.  The icon reacts and there is a brief pause but that's it.

I believe that the only difference between the way I built the bottom and top covers is that for the top covers I referenced the top edge of the rear chassis to create the XY working plane for the initial sketches.

Regards,
--Steve

[Chassis unfold example 180123.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1656399/Chassis.unfold.example.180123.FCStd.zip)

shrines, 2018-01-23:
Hi Shai, I figured out the cause of the problem with the part.

I went back and step by step deleted the various steps used to build the sheet metal part until I was able to unbend the part.  Then I redid the build sequence and checked at each step to ensure I could still unbend the part.

I discovered that the problem most often comes about during the creation of a bend with a non-zero value for one of the gaps.  I cannot tell which side of the bend Gap1 or Gap2 refers to.  My motus operadi is to guess and look at the result to see if I guessed correctly.  If not, I undo and try again.  When I am right in my guess then the part continues to build correctly.  When I am wrong, undo and try again then the part is broken.  The fix is, rather than to undo, is to delete the offending bend and create it over again.

The impression that nothing happens when I press the unfold icon is a result of the part being broken early enough in the part build sequence that no shape is created during the unbending.  

From a user perspective, I would break less parts if I had some indication of which edge of the Bend Gap1 or Gap2 referred to.  

For now I will just unbend after every bend with a gap to make sure that I haven't broken the part.

--Steve

ulrich1a, 2018-08-19:
[image]
The gap option 1 produces an additional relief at the other side, which breaks the unfold operation. This should be avoided.

Ulrich
[gap_problem.fcstd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2300382/gap_problem.fcstd.zip)

luzpaz, 2019-07-09:
@shaise probably can close this ticket as it is not related to the WB but the model..right?

shaise, 2019-07-09:
I agree. closing.

closed 2019-07-09 by shaise

shrines, 2019-07-12:
I agree, too.