shaise/FreeCAD_SheetMetal #22: Propose to add one more function: convert solid body to sheetmetal

naked-boy, 2018-01-29:
This sheetmetal workbench is very nice to me. It enables users to create sheetmetals, which is commonly used in industry production.
But currently there is only one way to create sheetmetal geomety in this workbench: to add sheetmetal geomety at an existing edge. In some circumstances, it is not convevient. Product below is an example.
[image]
I'd like to propose to implement one more function: to convert solid body to sheetmetal. This is not anything new. We can see this function in any main stream software like SolidWorks. Please kindly consider it. Thanks a lot.

normandc, 2018-07-28:
What you are asking for can be very complex to implement. What if your solid body is a shape that cannot be unfolded: do you want this tool to tear up corners? Replace hard edges with rounded bends? I can't fathom the complexity of programming this.

Maybe you don't know this, but you can already unfold a solid body if it can be unfolded, meaning it must follow rules like: uniform thickness, existing bends, etc. This model can be made with any existing Part or PartDesign tool, or even imported.

Your screenshots are not complete and don't tell us the complete shape, but I'm pretty sure it can be unfolded.

EstebanFPV, 2024-11-27:
even a basic version of this could be quite useful. the problem comes as soon as you start dealing with angles. if you model a part with a loft, say, the top and bottom are not 90 degrees to the sheet, and the unfold fails. for instance, if you have a shape that has 16 sides on the bottom lofted vertically to a square. this makes an approximated square to round transition that can be cut in 2 halves and bent with a press brake. before being welded together with flanges top and bottom. To model it with the current capabilities of the workbench is tedious at best, for more complicated shapes, nigh impossible