shaise/FreeCAD_SheetMetal #23: Unintentional face added by bend

shrines, 2018-02-01:
Hey Shai,

In the attached example, I put in a 0.07 inch gap at one side of the "Cover top right middle bend."  When I did so a 0.07 inch wide face was also incorrectly added to the front left edge of the part next to the middle bend.  Thereafter, when I add the "Cover top right front bend" the 0.07 inch face is not included in the bend, thus resulting in an unintended gap at the front left corner.
 
The effect is repeatable with a new part. by creating one sketch, one pad and one bend with gap.

--Steve
[Unintentional face added by bend.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1686139/Unintentional.face.added.by.bend.FCStd.zip)

shrines, 2018-02-01:
Shai,
A bit more information: The unintended face aligns with the back surface of the cut associated the gap edge.
--Steve

shaise, 2018-02-01:
Hi Steve,

Yes, I see what you mean. I'll see what I can do.

shaise, 2018-02-01:
Fixed!

closed 2018-02-01 by shaise