shaise/FreeCAD_SheetMetal #24: creating a gap breaks the bend on freecad 0.17 body

ouinouin, 2018-02-17:
Hi, 
while bending with no gap from a body (Part design next) : no issues as shown on picture below
[image]

while creating a bend with a gap, whatever the orientation i got this : the direction of the fold created is going backwards (may be it has to do with local coordinates of the Part vs global ccordinates of the document ? 
[image]

while doing the bend from the part workshop (first plate crated from a cube), i dont experience the issue : 
[image]

OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13247 (Git)
Build type: None
Branch: master
Hash: 5022c8df5cef0690f2c8acff4bcebd82655a5f04
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: French/France (fr_FR)

shaise, 2018-02-17:
can you attach a sample file? (FCStd)

ouinouin, 2018-02-20:
Hi, here is an example, I looked into your code , (despite not being coder) I can guess this has to do with the normalAt function and its use, there is interreseting things in the freecad forum about this but I still dont understand the U,V parameters it takes (freecad documentation is wrong  saying it takes a float : it take a couple of floats.)
here is the file
http://ouinouin.net/nextcloud/index.php/s/GrF14XfX91CQzeK

shrines, 2018-02-25:
Hey Shai,
I just experienced the same problem on a relatively simple metal box.  

I initially assumed the problem was associated with the naming problems that occur when a new face is created because a gap is modified.  I therefore rebuilt the attached example without any deletes or redos in the build sequence.  The problem reoccurred immediately after inverting the bend and specifying a gap.

[chain box 180223.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1755532/chain.box.180223.FCStd.zip)

Regards,
--Steve

shaise, 2018-02-25:
Hi Steve,

Thanks for the input. I am extremely busy now with my "real" job, but hopefully I'll find time for it soon.

shai

shrines, 2018-03-03:
Hey Shai,
Another data point: I just opened the example chain box file and changed the radius on the broken bend from the default 0.39" to 0.01".  The bend repaired itself.
--Steve

shrines, 2018-03-05:
Hey Shai,
After rebuilding the part a couple more times I think I have a repeatable phenomenon.  The bend breaks after inserting a gap only on parts that have been built using another part's face as an initial (independent) position reference.  If I build the part using exactly the same sequence but without referencing another part face then the part builds correctly.
--Steve

shaise, 2018-03-06:
Steve,

I just made some tests, and it seems something has changed in FC system. when I use older version of FC0.17 (build 12920) the bends are built correctly. But I have now way of telling which exact build broke it, so it is hard to track the reason

shai

shaise, 2018-03-06:
Steve, 

Well, this is strange. I downloaded the latest FC (13662) And it works correctly as well. I just loaded your file, marked the problematic bend004 for recompute and when I refreshed it was fixed.

Can you do this test?

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13362 (Git)
Build type: Release
Branch: master
Hash: 8c6b4ad1a60e2af6c1512cd01b6f47620612a5bd
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

shrines, 2018-03-06:
Hi Shai,
I'll download ver 13362 and try it with the example file later today.
--Steve

shrines, 2018-03-06:
Shai,

I opened the model in 0.17.13662 and saw the same bend correction on
recompute as you did.  I then added a couple of additional bends and had
the problem reoccur.

See the attached.

--Steve

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13362 (Git)
Build type: Release
Branch: master
Hash: 8c6b4ad1a60e2af6c1512cd01b6f47620612a5bd
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

shrines, 2018-03-06:
Interesting: When I opened the model a second time and marked the bend for
recompute it healed itself.

shaise, 2018-03-08:
This is a strange inconsistency. I wonder if it happens also when not reopening. Can you try reproducing the issue, then immediately without closing the file, recompute?

shrines, 2018-03-08:
Shai,

This morning recreated the sequence, as follows:
1.  I opened the chain box 180223 example file, verified that the incorrect
bend was visible, marked the bend for recompute and confirmed that the bend
direction was corrected.
2. I then deleted all bends on the top part and started rebuilding it.  I
was able to recreate the error on the second bend.
3. I marked the incorrect bend for recompute.  Recomputing had no effect on
the incorrect bend.
4. I saved and closed the chain box file as is, then reopened it, confirmed
the bend was still backwards, marked it for recompute, recomputed and the
bend corrected itself.

--Steve

shaise, 2018-03-08:
Thanks, Steve.
I now need to find time to debug it.

shaise, 2018-03-08:
Steve,

I think i fixed the issue (looks like rounding errors). Can you please test and confirm?

shai

shrines, 2018-03-08:
Shai,

It looks like you fixed the initial wrong direction bend problem but
introduced a new oddity.

I verified that your update fixed the bend direction and then started
adding new bends to see if I could break it.  No problems with bends that
bent the entire edge. When I added a gap to a trial bend the entire part
rotated 90 degrees on the X axis, changed thickness and generally broke
(see attachment).  It is  possible to delete the defective bend and the
model remains intact.

--Steve

shaise, 2018-03-08:
Hi Steve,

Did you attache anything?

shrines, 2018-03-08:
Oops.  Here it is.

shrines, 2018-03-08:
(I forgot t to add the .zip to the name to get through the filters.)

easyw, 2018-03-08:
@shrines 
if you post to github with a mailer, GH drops the attachments

shrines, 2018-03-08:
Okay, I'm back from San Diego and working from my real desk.. Here's the example file built from your build.
[chain_box_example 180308.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1795068/chain_box_example.180308.FCStd.zip)

ouinouin, 2018-03-08:
opened the chain_box_example 180308.FCStd example, removed the bend 8 from the body named "top" and tried another bend on your part, i got a also a bad result.
[image]
the preselection in yellow is the face I tried to bend.

on my side I did some tests and the first bug seems to have been cleared even while putting the initial body in some strange positions (not normal to the X Y or Z plans).
OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13375 (Git)
Build type: None
Branch: master
Hash: 2bfea410fce1434fec0f853b032549aa2cd3c47e
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: French/France (fr_FR)

shrines, 2018-03-10:
Hey Shai,

I think you might find the attachment interesting.  I created two identical parts with just two bends each.  On the first part I included gaps in the first bend and on the second part I did not.  The direction of the second bend is incorrectly reversed on the part that has gaps in its first bend.  

I'll bet this phenomenon is at least partially related to your flipped bend direction problem.  It might also explain my previous confusion as to which end of a face gap1 applies to when building complex bent parts.

[Simple bend Examples.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1799702/Simple.bend.Examples.FCStd.zip)

--Steve

shrines, 2018-03-10:
A bit more on the bend direction change:  The bend only changes direction when the gap of the first bend is "next to" the second bend.  In the attached example Body000 Bend000 gap1 and gap2=0.01; Body001 Bend002 gap1 and gap2 =0; Body002 Bend004 only gap1=0.01; Body003 Bend006 only gap2=0.01.  The second bend only reverses direction on Body000 and Body002 where gap1 (the gap between bends) is non-zero. 

[Four Simple bend Examples.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1799722/Four.Simple.bend.Examples.FCStd.zip)

From my naive perspective that would narrow the problem down to lines 109-121 in SheetMetalCmd.py
--Steve

shaise, 2018-03-11:
ouinouin,

This problem is not sheetmetal related, or have anything to do with the fix. Here is how sheet metal works when you make a bend: It creates 2 seperate solids - a bend and a wall, then fuses them to the main body. The bend and the wall are created correctly:
[image]

After fusing all solids the error happen:
[image]

OCC is known to have issues with flat face to face fusing of objects.

shaise, 2018-03-11:
Steve,

the bend direction is also somewhat uncontrollable. It is randomly based on the direction of the edges of the face that bends. When you add a gap, the geometry changes (more edges then without gaps) and quite possibly the edges might point to a different direction.

shai

ulrich1a, 2018-08-19:
> OCC is known to have issues with flat face to face fusing of objects.

The unfold tool needs also to generate new geometry. I do not using the OCCT fusion but do collect all faces and build a new shell and than a new solid. This approach is more robust in OCCT. 

Ulrich

luzpaz, 2021-02-11:
What's the status of this ticket?

luzpaz, 2021-04-13:
@shaise bump

shaise, 2021-04-14:
hi @luzpaz ,

I have reconstructed the part under freecad 19 (just went bend by bend, and select the correct face) and I see no issues.

shai

luzpaz, 2021-05-15:
@shaise close ticket then?

shaise, 2021-05-15:
maybe...

luzpaz, 2022-11-09:
Closing ticket

closed 2022-11-09 by luzpaz