shaise/FreeCAD_SheetMetal #25: Unbend fails when bend abuts inner corner

shrines, 2018-03-03:
Hey Shai,

I simplified the Issue 9 example part down to one L-shaped part with two bends and discovered a (perhaps related) problem.  The two bends are both associated with the inner corner of the L. One bend has a gap to keep the two bends from interfering with one another when unbent.  

The part is drawn correctly after both bends.  However, it is not possible to unbend the part starting from the bend face that abuts the inner corner.  You can unbend the part if you start from the other bend face (the one with the gap).  In that case the part unbends but the abutted bend breaks during the unfold.

It may be that you have to add a cut at the inner corner of the L to isolate the face edge before you initiate the bend.

[Chain box cover.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1777950/Chain.box.cover.FCStd.zip)

Regards,
--Steve

luzpaz, 2019-07-17:
@shaise care to weigh in?

shaise, 2019-07-20:
The unfold will never work since the fold is kind of 'illegal'. When you make a fold completely to the edge, it will fuse with the main body, creating an unfoldable model. The solution is to add a small gap (like 0.1mm) and, if a relief is unwanted, set reliefw to 0.0. This is also a more realistic model, as the laser cutter leaves an inherent gap anyways.
Attached model for reference

[Chain.box.cover.fixed.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/3413639/Chain.box.cover.fixed.FCStd.zip)

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]