shaise/FreeCAD_SheetMetal #27: missing WB icon in SVG format

easyw, 2018-03-22:
Hi @shaise 
it would be possible to have the main WB icon as svg?
(D:\shai\FreeCAD\mkwall_xpm)
In HDPI monitors the actual XPM is little if used inside Selector Tool...
[image]
Thx
Maurice
PS if you could share the svg file, I could make a PR to switch from xpm to svg 😄

shaise, 2018-03-22:
Hi Maurice

I've been thinking of changing them for quite some time... Thanks for giving me the incentive. Can you add these Icons and generate PR?
[WBLogos.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1837537/WBLogos.zip)

easyw, 2018-03-22:
done a PR https://github.com/shaise/FreeCAD_SheetMetal/pull/28 😄 
PS do you also have the old SMLogo in SVG?
Thx
Maurice

shaise, 2018-03-22:
Of Course!

Attached. Funny thing is that it was just a trial sketch when I first started SM workbench, and it stayed...
[SMLogoOld.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1838071/SMLogoOld.zip)

closed 2018-03-22 by easyw

shaise, 2018-03-22:
BTW, how do you activate this tool bar? looks more convenient then the drop box.

shaise, 2018-03-25:
After considerations, mainly because the new icon does not look good in small size, I reverted to the original Icon with your nice sheen tweak... thanks!