shaise/FreeCAD_SheetMetal #29: Mitered bend unfold error

shrines, 2018-03-27:
Hey Shai,
I just updated to your latest sheetmetal workbench and decided to try out mitered bends.
Pretty cool!
I did run into a problem, though.  If I make one mitered bend the part unfolds exactly as expected.  If I make a second bend from the face of the mitered bend then the part won't unfold.  Instead I get one of two error messages, depending on which face I select to initiate the unfold.

[Mitered bend.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1853505/Mitered.bend.FCStd.zip)
[image]
[image]

Regards,
--Steve

shaise, 2018-03-28:
Hi Steve,

Unfortunately the unfolder code is not mine, and it's rather complicated for me to understand it. It belongs to the very talented Ulrich Bramar. 
can you try his latest macro from here: https://forum.freecadweb.org/viewtopic.php?t=15421#p122997
I'm pretty sure I have updated to his latest version, but I have already heard that sometimes the original works as opposed to the integrated one.
If it does not work, try posting a question in the forum.

shai

ulrich1a, 2018-08-19:
The problem with the part Mitered_bend is, that the flat sheet with the miter angle has bspline edges and not straight lines (Edges 20 and 16). The following bend (bend001) now produces a <SurfaceOfRevolution object> instead of an <Cylinder object>. The <SurfaceOfRevolution object> is not supported by the unfolding script. It is not planned to support this type surface, as it has no defined radius.

Ulrich

luzpaz, 2019-07-09:
@shaise Maybe label this as "Won't Fix" ?

shaise, 2019-07-09:
Agree. how?

luzpaz, 2019-07-09:
@shaise the same way you made and 'enhancement' label you make one for 'wont fix' and apply it to this ticket. Then close the ticket.

shaise, 2019-07-09:
thanks. closing

closed 2019-07-09 by shaise