shaise/FreeCAD_SheetMetal #30: Set K-factor for unfolding

iadenisov, 2018-03-30:
During the unfold procedure the length block should be dependent from the K-factor. However I do not find the way how to change. Such setup parameter will be extremely useful.

shaise, 2018-03-30:
iadenisov,
I definitely agree. Unfortunately the unfold is not my code, I just integrated Ulrich Bramar's Unfold macro. Try leaving a post in the freecad forum.

shai

iadenisov, 2018-03-30:
I found that author also knows that it is required. 

SheetMetalUnfolder.py
```
# To do: 
# - handle a selected seam
# - handle not-circle-curves in bends
# - detect features like welded screws
# - make a view-provider for bends
# - make the k-factor selectable
# - upfold or unfold single bends

## https://forum.freecadweb.org/viewtopic.php?t=15421#p122997 
 
```

shaise, 2018-06-25:
K-Factor manipulation added to ver 0.1.30

closed 2018-06-25 by shaise