shaise/FreeCAD_SheetMetal #32: [FR] Allow editing the source face for Bend or Extend feature

normandc, 2018-05-06:
It would be useful to re-attribute a bend or extend feature to a different face ID after its creation. Due to FreeCAD's [topological naming](https://www.freecadweb.org/wiki/Glossary#Topological_Naming) issue, if changes are made to a bend's properties (like gap1 and gap2) that changes the number of faces it generates, then following operations will likely fail or produce unexpected results after a recompute. An editing option would allow the end user to repair failed features.

This is a recent example where editing bends would be useful: [FreeCAD forum - Sheet Metal bending](https://forum.freecadweb.org/viewtopic.php?f=3&t=28572)

Thanks.

shaise, 2018-05-07:
I Agree this is a good feature. I will take a look at it as soon as I have time.

shaise, 2018-06-25:
Done on version 0.1.20

closed 2018-06-25 by shaise

normandc, 2018-07-28:
Thank you! I was trying to find this in the Data tab, but just now realized that you only need to double-click on the Bend feature in the Model tree to open a dialog.

normandc, 2018-07-28:
I just noticed, following issue #37 that if I have a model with Bend, Bend001 and Bend002, and I delete Bend001, then Bend002 fails. I was expecting to be able to edit Bend002 and reattribute a face to it, but I'm getting the following error:

`Traceback (most recent call last):
  File "/home/normand2/.FreeCAD/Mod/sheetmetal/SheetMetalCmd.py", line 373, in setEdit
    taskd.update()
  File "/home/normand2/.FreeCAD/Mod/sheetmetal/SheetMetalCmd.py", line 429, in update
    if isinstance(f[1],list):
<type 'exceptions.TypeError'>: 'NoneType' object has no attribute '__getitem__'
Exception (Sat Jul 28 13:05:15 2018): getEditDialog() not implemented`

shaise, 2018-07-28:
when you delete a parent bend the next bend have no base object to work with, hence I guess it fails.