shaise/FreeCAD_SheetMetal #33: bending machine wants to have the middle of the bend marked

easyw, 2018-05-07:
Hi @shaise @ulrich1a 
following this thread
https://forum.freecadweb.org/viewtopic.php?f=3&t=28331#p229621
it seems the SheetMetal WB should generate a sketch of middle of the bend.
It would be very useful to have also a similar option implemented.
Thanks for any insight.
Please find attached a simple FC file with the two sketches compared (one from the unfolded and one from the suggested middle references)
[unfolding-FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/1979391/unfolding-FCStd.zip)
[image]

shaise, 2018-06-25:
sketch + bending marks added to ver 0.1.30

closed 2018-06-25 by shaise

easyw, 2018-09-17:
Hi @shaise 
I managed to obtain aSketch for the Outline of the unfolded part.
Please have a look at:
https://github.com/easyw/FreeCAD_SheetMetal/tree/test
(you need to check the option to have bending line sketch to trig this feature)
I haven't added the option to select a different color for the Outline but it can be easily added.
I tested the tool with some of the old critical file and it seems fine ...
Later, if this will be merged, it could be an easy way to remove the outline geometry from the 'Unfold_Sketch'
[image]

easyw, 2018-09-17:
@shaise 
I've added also a first attempt on Internal geometry sketch generation.
This is definitively to be improved. (like adding a small tolerance on geometry comparison). 
Moreover also OutWire Sketch should have always the placement of the main Sketch.
(tested with this file [2ways.FCStd](https://forum.freecadweb.org/download/file.php?id=36894) )
found here https://forum.freecadweb.org/viewtopic.php?f=10&t=11303&start=60#p171492

shaise, 2018-09-17:
the latest version already generates sketches with selectable colors. Did yo test it?

easyw, 2018-09-17:
@shaise 
yes, but I haven't added this option to the new sketches I'm creating ('Unfold_Sketch_OutWire', 'Unfold_Sketch_Internal').
Please have a try with my test branch and see what is generating.
here a test file within all the new sketches.
[2ways-w_sketches.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2388680/2ways-w_sketches.zip)

easyw, 2018-09-18:
@shaise 
now the routines are much cleaner and the sketches are in line with previous code.
Still I would need to generate the sketches aligned to the others in every face selection case...
Would you mind to have a look at the code?
Thx
Maurice

easyw, 2018-09-18:
ok, I've managed to fix this...
I'm going to make a PR