shaise/FreeCAD_SheetMetal #37: Deleting bend cause errors

mcavallerin, 2018-06-28:
Hi,

I really appreciate your WB. I think you made a great job. In any case, would it possible in the future to consider that when deleting a bend the model doesn't show errors and recompute properly?
I mean, if you create bend  & bend001, if you delete bend bend001 gives error and you can't edit.

Is it because it has to rename faces?

I tried the same on partDesign and if a feature is not related to another, the model recompute properly.

Thanks.

Cheers.

shaise, 2018-06-28:
if bend001 is the last action, it should not generate an error. If you continue working on the model, then later delete a bend it might not recompute. and yes this is because of face renaming. This is an inherent issue in freecad

normandc, 2018-07-28:
In PartDesign, you can circumvent the topological naming issue by not linking sketches and features to faces on the solid. In SheetMetal, this is unfortunately impossible, because you need to select faces to create Bend and Extend features.

klausz, 2019-09-27:
Not sure if I miss something but the issue I have was that I can't change measurements within a drawing consisting of bends. 
So I decided to create a macro. Even there when changing parameters it seems the face name change. 

Having a look at the general logic (hopefully the code did not get destroyed when I simplified and removed double entries etc) having a center steel plate boy (which has a top and a bottom and exactly 4 faces) : 
..
`Gui.Selection.clearSelection()`
`Gui.Selection.addSelection('Unbenannt','Body','Bend.Edge9',3.5,96.4803,-2.5)`
`Gui.runCommand('SMMakeWall',0)`
`Gui.Selection.addSelection('Unbenannt','Body','Bend001.Face5',1488,384.858,0.662658)`
`FreeCAD.getDocument("Unbenannt").getObject("Bend001").radius = '2.5 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend001").BendType = u"Thickness Outside" `
`FreeCAD.getDocument("Unbenannt").getObject("Bend001").length = '215 mm' `
`Gui.Selection.clearSelection()`

`Gui.Selection.addSelection('Unbenannt','Body','Bend001.Edge10',1489.5,333.014,215)`
`Gui.runCommand('SMMakeWall',0)`
`Gui.Selection.addSelection('Unbenannt','Body','Bend002.')`
`FreeCAD.getDocument("Unbenannt").getObject("Bend002").angle = '0 deg' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend002").length = '23 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend002").miterangle1 = '25 deg' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend002").miterangle2 = '25 deg' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend002").AutoMiter = False `
`Gui.Selection.clearSelection()`

`Gui.Selection.addSelection('Unbenannt','Body','Bend002.Face18')`
`Gui.runCommand('SMMakeWall',0)`
`Gui.Selection.addSelection('Unbenannt','Body','Bend003.')`
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").invert = True`
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").length = '35 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").radius = '2.5 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").angle = '91 deg' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").extend1 = '0.1 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").extend2 = '0.1 mm' `
`FreeCAD.getDocument("Unbenannt").getObject("Bend003").BendType = u"Thickness Outside"`
`Gui.Selection.clearSelection()`

It seems its always going ahead from the previous parte added even when I am on another of the 4 faces. Further it seems I cannot remove the end, a part or the full branch added on one of the 4 different faces. 

But it could also be that i miss some logic in Freecad (how to reorganize these different Bends in a one line Bend (as pseudocode):  
`Bend .. Bend014` to a center with four linear branches. e.g. 
`Bend..Bend002;` 
`Bend..Bend004;`
`Bend..Bend008;` 
`Bend..Bend014` 
(where e.g. Bend 001 and Bend002 are only included in Bend..Bend002 and 
Bend..Bend004 has Bend+Bend003+Bend004.      
If not already there changing the tree structure and simplifying face access could improve the logic and flexibility.