shaise/FreeCAD_SheetMetal #38: How to unfold with one bending line instead of two lines?

ceremcem, 2018-08-17:
Currently, unfolding the following part: 

[image]

produces following output in DXF file (via TechDraw workbench): 

[image]

I added the red line with Gimp, which is what I wanted instead of lower and upper bending lines (because I'm bending the sheet metal with air bending technique). 

How can I produce one bending line instead of these two?

shaise, 2018-08-17:
I do not know if you have the latest version, but in the latest version the unfolder creates a "projection sketch" with only one line in the middle, ready for bending machine.
(when you press the unfold button, you get a task dialog. make sure the "create projection sketch" is selected)

easyw, 2018-08-17:
If you post the file I could have a look.

ceremcem, 2018-08-17:
Absolutely, [here it is](https://github.com/ceremcem/fc-asm3-test/blob/4c2718f9e69ea58e55085fafe95d03121dd50f71/assembly.fcstd)

easyw, 2018-08-17:
@ceremcem 
you are trying to unfold a link to your object.
A3 branch as a ton of new features that are not covered by FC official WBs as like the external WBs.
SheetMetal doesn't support Links ATM.
This is normal, considering you are working with an experimental dev branch...

If you unfold the original object (or even a simple copy of it) SM will create correctly the unfold sketch.
[image]

ceremcem, 2018-08-17:
Thank you! 

The main reason for trying to unfold the linked object is that I need to both see the 3D drawing and the unfolded version. As I was only using the `unfold=true` property, I thought that would be only possible with unfolding a linked object (since I needed to track changes (such as cutting windows, etc) on the parent). 

Anyway, since unfold tool displays both folded and unfolded versions, there is no need to create the linked one.

closed 2018-08-17 by ceremcem