shaise/FreeCAD_SheetMetal #41: Changing bending line color by default

ceremcem, 2018-09-11:
For helping mechanical operations, it would be nice to have bending lines in different color on unfold.

shaise, 2018-09-11:
I Agree. Unfortunately freeCad sketch has no option for separate line colors.

easyw, 2018-09-11:
it would be nice to separate the sketches like in commercial sw:
1) the unfold external outline
2) the fold lines
3) the holes/internal cuts
then different colors can be handled too.
https://forum.freecadweb.org/viewtopic.php?f=21&t=30000#p247481
https://forum.freecadweb.org/viewtopic.php?f=3&t=28331&p=241213#p229724
https://forum.freecadweb.org/viewtopic.php?f=3&t=27928&p=252918#p252891

shaise, 2018-09-11:
I thought about this solution. (it is similar to what I did on Path Simulator)
I did not want to generate too much objects on the tree view. However, perhaps I will add a check box to the unfolder task letting the user decide if to generate separate sketches.
This will at least separate the bend lines. For the internal features to be separate I need Ulrich's help.

shaise, 2018-09-15:
Done. 
(At least for bend lines)

ceremcem, 2018-09-15:
Where do we set "bend lines color"?

shaise, 2018-09-15:
Right now there is no way. Is this a real need?

shaise, 2018-09-15:
OK. 
Added color selection as well.

ceremcem, 2018-09-16:
To remove any misunderstanding, I meant the following lines:

[image]

shaise, 2018-09-16:
When you open the unfold task now, you see the colors on the side and you can change it.

ceremcem, 2018-09-16:
Now I see it, thanks! (FreeCAD warns users that FreeCAD should be re-opened after a first WB install, but it doesn't warn for WB updates. That might be the cause)

closed 2018-09-16 by ceremcem

ceremcem, 2018-09-16:
Unfortunately changed line colors are not meaningful until the TechDraw preserves the line colors while generating projections: https://github.com/WandererFan/FreeCAD-TechDraw/issues/64

shaise, 2018-09-16:
Yes, I agree.