shaise/FreeCAD_SheetMetal #44: Please consider using GitHub tags and releases

luzpaz, 2018-09-15:
https://stackoverflow.com/questions/18216991/create-a-tag-in-github-repository

luzpaz, 2018-09-15:
Honestly you already have one. You'd just tag any minor version changes and tag and  release major version changes.

shaise, 2018-09-16:
Great. Will do.

ceremcem, 2018-09-18:
Also, would you consider using Github tags? Like `bug`, `feature request`, `discussion`, etc...

shaise, 2018-09-20:
I don't really understand what is a meaning of tagging a version as a 'discussion' or 'bug'

ceremcem, 2018-09-20:
You can add such labels to the issues:

[image]

...which would be quite informative for us and newcomers. You can use the tool on the right side inside an issue: 

[image]

...and of course, create new ones: 

[image]

shaise, 2018-09-20:
OK I see, this is not a tag, it is a label, and used for issues. Makes sense now.

ceremcem, 2018-09-20:
Oh, my mistake!

luzpaz, 2021-04-08:
Request, adding a little more consistency. Last tag would be very much welcome

[image]

luzpaz, 2021-07-03:
@shaise can we start doing this?

shaise, 2021-07-04:
what needs to be done here?

luzpaz, 2022-11-09:
@shaise the only thing is to also start using releases and release tags (https://github.com/shaise/FreeCAD_SheetMetal/issues/44#issuecomment-815740954). Would you be in to that ?

shaise, 2022-11-10:
So every time I update the version, add a tag as well?

luzpaz, 2022-11-10:
Yea, makes sense to me.

shaise, 2022-11-10:
will do.

closed 2022-11-10 by luzpaz