shaise/FreeCAD_SheetMetal #52: 'Relif' is a typo?

luzpaz, 2019-01-19:
https://github.com/shaise/FreeCAD_SheetMetal/blob/master/SheetMetalCmd.py#L650-L651

luzpaz, 2019-01-19:
[image]

What's the hard in just changing all instances of it in the source, committing it to a separate branch, running it and seeing if it crashes? 
[image]

shaise, 2019-01-19:
It will not crash, but if I load an old file that was saved with the wrong spelling, It will not load.

shaise, 2019-01-20:
OK, there are two options:
1. change most typos to Relief, but leave internal property names to relif, this will be compatible with old files, but will display correctly for the user.
2. change everything to Relief. This will not crash when loading old files, but any editing to sheetmetal featuers will reset relif values, and will display double features on the feature list (only for old files).

What do you think is best?

luzpaz, 2019-01-20:
Fast method is 1
Smart method is 2
Is there a way to write some code that when encountering the misspelling could auto-change it ?

shaise, 2019-01-20:
Theoretically it is possible to correct, but then you will have "relif" in the code since you have to ask if a parameter is "relif" then change to relief.... I think its an ugly patch just for typo.
method 2 is easy to do, but if you load old files, you will see 2 parameters in the properties list: both relif and relief, so it is ugly ad well....
The first method means only internal parameters (at 4 places) will be relif but it will not be visible to the end user. 
I can do any of the 3 you like.

luzpaz, 2019-01-20:
I'm for being `future friendly`...so I'm for method 2 if people can quickly delete the `relif` parmeter. Eventually, the problem will disappear as more people learn and use this Workbench.

shaise, 2019-01-20:
Friendly it is!
Done.

closed 2019-05-14 by luzpaz