shaise/FreeCAD_SheetMetal #54: Unfold option Issue

jaisekjames, 2019-01-23:
See Issue

https://forum.freecadweb.org/viewtopic.php?f=22&t=25281&start=150#p279502
please change from
unfoldLength = ( bendR + kfactor * thk / 2.0 ) * bendA * math.pi / 180
to 
unfoldLength = ( bendR + kfactor * thk / 2.0 ) * bendA * math.pi / 180.0
Add decimal place to 180.

shaise, 2019-01-24:
Thanks! I will take a look

shaise, 2019-01-24:
Hi,

The 180.0 is not the problem. The problem is difference between European and American standard of k_facror (one of them is 1/2 the other). I have reverted it to the European standard, which is compatible with Ulrich's original unfolder script. Now both unfolding from your system, and Ulrich's system match.

f3nix, 2019-01-24:
Thanks! :)

I have tested the changes and here are my observations.

Cases 1 and 3 work as expected now. :+1: 
I have a question about case 2. Should it use the kfactor set in the kfactor parameter (property).

[image]

It seems it uses 0.65 in this instance.

`up Face7 k-factor: 0.65`

Cheers,
Mateusz

shaise, 2019-01-24:
The k factor in the properties are total different system and works only on sheetmetal prefolded using the sheetmetal WB. If you make a sheetmetal model using another system, you can only unfold it using the unfold button. The unfold button uses its own K-factor, and its the same factor for all folds. The properties system can set a different k factor to each fold.

f3nix, 2019-01-26:
Ok. Here is what I'm doing.

1. Create body and sketch.
2. Draw a rectangle (82mmx90mm).
3. Pad it (4mm).
4. Create a bend.
5. Change default properties:
   * length: 82mm
   * radius: 4mm
   * kfactor: 0,86
   * reliefd: 0mm
   * reliefw: 0mm
6. Select face and click "Flatten folded sheet metal object".

The unfolded length is **182,33mm** (which is different than other two unfold cases).

Report view says:
`up Face7 k-factor: 0.65`

It does _not use_ the **0,86 kfactor** that I have set in the bend property.

Is it the intended behaviour or am I doing something wrong?

Here is the example file: [L-bracket-final.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2799591/L-bracket-final.zip)

```
OS: Debian GNU/Linux buster/sid
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15703 (Git)
Build type: Debug
Branch: master
Hash: 80f82b59db1bae91115921d7da3190f855e1041f
Python version: 2.7.15+
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Polish/Poland (pl_PL)
```

Cheers,
Mateusz

shaise, 2019-01-27:
Sorry if I wasn't clear,

The unfold button disregards ALL fold properties, and works with its own k-factor system.
The only way to use the k-factor in the the properties, is to set the unfold to true in the properties as well. This means, and I know its tedious, that if you want to unfold the entire object using the k-factor in the properties, you need to go fold by fold and set the unfold property to true as well.

f3nix, 2019-01-30:
Thanks for clarifying. I think this issue has been resolved.

Thank you again! :)

Cheers,
Mateusz

ceremcem, 2019-04-20:
> The unfold button disregards ALL fold properties, and works with its own k-factor system.
The only way to use the k-factor in the the properties, is to set the unfold to true in the properties as well. This means, and I know its tedious, that if you want to unfold the entire object using the k-factor in the properties, you need to go fold by fold and set the unfold property to true as well.

Wow! I was about to open an issue about "k-factor is not regarded while unfolding" and I just saw this explanation. What is the use case of this behavior? Why isn't the k-factor from the bend properties simply used while unfolding?

ceremcem, 2019-04-24:
@jaisekjames I now think the issue is solved.

luzpaz, 2019-07-09:
@shaise can this ticket be resolved?

shaise, 2019-07-09:
yes, I think it can be closed

closed 2019-07-09 by shaise