shaise/FreeCAD_SheetMetal #55: Changing a Parameter breaks the model

dmun19, 2019-02-05:
Hi !

Freecad 0.17 (and 0.18) on Windows 7 64 Bit.
I use a spreadsheet where I specify the parameter of a metal chassis like width or high, which used to work fine with an old version of the sheet metal workbench but with an up to date version changing a parameter breaks the model. 

Attached the freecad file (freecad sheetmetal error.zip) open it in Freecad with and uptodate version of the sheetmetal workbench installed, open the spreadsheet and change the parameter "High" from 120 to 200, freecad will recalculate the chassis and raises and error (see model tree on the left)

No install the old sheetmetal workbench (file: sheetmetal.ok.zip), restart freecad and do the same steps again and now everything works (hopefully) fine.

I have tested this issue on two PCs, always with the same result. The old workbench does the job the new one fails.

I hope this is clear otherwise let me know and I will try to give more information. The sheetmetal workbenach is a great tool and I really do not want to miss it.

Thank you. Dirk 

[freecad sheetmetal error.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2834042/freecad.sheetmetal.error.zip)
[sheetmetal ok.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2834037/sheetmetal.ok.zip)
[sheetmetal broken.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2834038/sheetmetal.broken.zip)

shaise, 2019-02-07:
did you generate the original file on old version of sheetmetal?

dmun19, 2019-02-07:
yes, with the version (or older)  that I have attached to the previous post

shaise, 2019-02-07:
OK. This explains it. There is a fundamental issue with freecad naming of faces. Changing parameters of base objects can only work if the face names stay coherent. When you use a new sheetmetal code, this can cause different naming of faces due to different methods of geometry generation. You can see what happens as follows - change W in the spreadsheet from 120 to 130, you will get the error you talked. this happens since Extrude001 is now on the wrong face. click Extrude001 in the tree and hit space to highlight it - you will see what I mean. 
Since its a fundamental freecad issue that can not be dealt, we added a feature that lets you reselect the correct face:
double click on Extrude001 on the tree. A dialog will appear with the face that this extrude is based on. Now in the model, select the correct face, and click the update button in the dialog. Click OK and now this face is corrected. You will need now to go up the tree one by one until you fix all. From now on this will work with the current sheetmetal WB, and changing parameters in the spreadsheet should work.

dmun19, 2019-02-07:
That's not good. Isn't there an other way to fix this problem ?

shaise, 2019-02-07:
Sorry, but no. As I told you this is a basic freecad issue that affects all workbenches
Here is an updated file that now works. you can change parameters and it will work.
Be sure to update your sheetmetal WB, I just added a needed fix.
[freecad sheetmetal error fixed.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2841479/freecad.sheetmetal.error.fixed.zip)

dmun19, 2019-02-07:
OK, I give it a try. Thank you !

shaise, 2019-02-07:
sorry, the sketch was also needed to be fixed. It was over constrained. 
[freecad sheetmetal error fixed.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/2841513/freecad.sheetmetal.error.fixed.zip)
attached

dmun19, 2019-02-07:
Thank you. The sketch was just an sample, I did not check it, sorry. Anyway, it looks like that I can not fix all problems the way you described it so maybe the best way would be to redraw the projects when I use them again to make sure that they are up to date. I hope such a problem will not happen again.

dmun19, 2019-05-25:
Hi, I am sorry but I must reopen this issue. Same problem again with Freecad 0.18. Changing a paramter of a model which was created with Freecad 0.17 and Sheetmetal in Freecad 0.18 with the latest Sheetmetal Workbanch will lead to a useless model. 
I have attached a sample file. The model was created with Freecad 0.17 and the latest Sheetmetal WB. Open it in Freecad 0.17, an change the Radius Parameter on the Spreedsheet and check the model. Works fine, no problem. No open the file with Freecad 0.18 and change the radius parameter on the spreadsheet and then check the result. The model is now useless and now way to go back.
Any idea what the problem is now ? Is this problem still caused by freecad naming problem ? 
By the way I made some tests with a prerelease of freecad 0.18 a few weeks ago and there was no problem.  It started with Freecad 0.18 16110 on Win7 64 bit, and 0.19 also is effected.
[Sheetmetaltest.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/3219703/Sheetmetaltest.zip)

luzpaz, 2019-07-09:
Is this a topological naming issue ?

shaise, 2019-07-09:
Indeed

luzpaz, 2019-07-09:
@dmun19 please read up on [Topological Naming](https://www.freecadweb.org/wiki/Topological_naming_problem) and how to avoid it

w0iy, 2020-03-14:
I'm not sure if I'm getting bit by the same bug. I'm a noob to the Sheet Metal plugin. I've made a 4 sided box and then made a rectangular hole in what is the front  panel. I used PART, made a Cube, then did Boolean-Cut. When I attempt to add a back panel, I select the edge and add a Bend. However, the new bend lip appears on a different edge than what I selected.

Using Freecad 0.17 and Sheet Metal 0.2.33

luzpaz, 2020-03-14:
> Using Freecad 0.17 

Why? Please use the stable version at least. Ideally you would use the 0.19_pre version

w0iy, 2020-03-15:
The issue is the same with 0.18 and 0.2.33
The results depend on the order of the operations.  If I make all the bends before doing the cut, I get the desired result. If I do the cut and then add the last bend, it is wrong.

luzpaz, 2021-03-03:
Try it on @realthunder's [LinkStage3 branch](https://github.com/realthunder/FreeCAD_assembly3/releases). He has solved the topological naming issue

ceremcem, 2021-03-03:
I'm only making addition to @luzpaz 's suggestion: If you use LinkStage3, then you can also benefit from  [SheetMetalUnfoldUpdater.FCMacro](https://github.com/shaise/FreeCAD_SheetMetal/blob/master/Macros/SheetMetalUnfoldUpdater.FCMacro), which automates updating unfolded geometries.

closed 2021-04-08 by shaise