shaise/FreeCAD_SheetMetal #67: Wrong output is generated for closed bent parts

ceremcem, 2019-04-13:
# Description 

If a bent part is closed on itself, start point of unfolded part is started determined erroneously. 

Here is a closed bent part: 

[image]

Here is the erroneous unfold:

[image]

# Workaround 

Breaking the loop with a negligible error fixes the issue: 

[image]

Here is the correct unfold:

[image]

# Attachment

Created with `LinkStage3@e9c60d4d`: [simple.fcstd.remove-me.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/3076025/simple.fcstd.remove-me.zip)

easyw, 2019-04-13:
@ceremcem 
you should need to state that you are using FreeCAD A3 realthunder branch.
This is obviously important in your user case ... Links are not implemented by FC main branch...

ceremcem, 2019-04-13:
I stated under the "Attachment" title with the exact commit hash but I understand that you mean stating under the "Description" title (in a more visible manner). 

Since this file doesn't use any LinkStage3 property (no linked documents, no Assembly Container, nothing), I preferred not to underline the LinkStage3 relation. Do you have a chance to reproduce such a drawing with FC main branch? Is it directly related with LinkStage3?

easyw, 2019-04-13:
you didn't test it in FC main branch did you?
The file cannot even be open cleanly in FC main branch.

ceremcem, 2019-04-13:
> you didn't test it in FC main branch did you?

No. 

> The file cannot even be open cleanly in FC main branch.

That's expected, obviously. However, what I asked is creating the drawing from scratch within FC main branch. It's a fairly easy drawing. Here is the steps: 

1. Create a rectangle with 30 mm wide with PartDesign WB 
2. Pad it 2 mm 
3. Add a bend at one side
4. Extend the bend eg. 50 mm. 
5. Add a bend
6. Extend the bend 30 mm 
7. Add a bend 
8. Extend the bend eg. 22mm
9. Add a bend to the other side of step 3
10. Extend that bend 28mm
11. Try to unfold this geometry

easyw, 2019-04-13:
so please create the example and post it here.

ceremcem, 2019-04-13:
@easyw You are so nice, thank you. Above steps can be applied to create the sketch from scratch under 2 minutes. I would appreciate if anyone could help the recreation the geometry before I need to download FC source and compile it from scratch.

easyw, 2019-04-13:
@ceremcerem
you are so nice  not taking your time to post a sample to work on.
Moreover nobody is asking you to build FC from source.
You should be smart enough to know there are plenty of FC appimages to be used.
It would take then less than the time you spent in your nice reply.

shaise, 2019-04-13:
This happens on main as well. it looks like the edge bends becomes fused as a single wall, hence the cut is not detected
[image]

[test_closed_loop_bends.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/3076333/test_closed_loop_bends.zip)

ceremcem, 2019-04-20:
@easyw 

> You should be smart enough to know there are plenty of FC appimages to be used.

Obviously I'm not :) I wasn't aware that FC main branch has appimages available too. I thought the images are only available for Assembly3. Anyway, thanks for letting me know that as this information saved my day today.

luzpaz, 2019-07-09:
FYI @ceremcem the [AppImage wiki page](https://www.freecadweb.org/wiki/Appimage) is auto-updated via script and has the most current appimage links to it. You can obviously also check https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre :wink:

luzpaz, 2021-04-06:
bump for status

alexneufeld, 2025-03-01:
This can be closed I think. The original issue of unfolding self-closed bent parts is a standard limitation of all sheet metal CAD systems I know of. The user can trivially work around this limitation by creating a rip/slit/seam somewhere on their part, which is needed for manufacturing anyway.

closed 2025-03-04 by luzpaz