shaise/FreeCAD_SheetMetal #68: Definition of k-factor

ceremcem, 2019-04-20:
Internet searches as well as [Wikipedia page](https://en.wikipedia.org/wiki/Bending_(metalworking)#K-factor) shows that the definition of "k-factor" is as follows: 

> K-factor is a ratio of location of the neutral line to the material thickness as defined by t/T where t = location of the neutral line and T = material thickness. (...) and usually between 0.3 and 0.5. 

Based on this information, [latest change of k-factor in SheetMetal](https://github.com/shaise/FreeCAD_SheetMetal/commit/62c40b14fe6fea982b0b90c0490ea417b79cef51) is misleading, thus should be reverted.

Moreover, the formula [`newNode.k_Factor = (0.65 + 0.5*math.log10(innerRadius/self.__thickness))`](https://github.com/shaise/FreeCAD_SheetMetal/blob/bf2c0c1ecbda40ceedda646c1fdf63b39810c896/SheetMetalUnfolder.py#L828) has no explanation. It should be explained or removed. There is a long discussion about this formula in SW forum and seems that it's still unclear: https://forum.solidworks.com/thread/38306

shaise, 2019-04-21:
Hi Ceremcem,
Yes... This issue game me a headache for quite sometime... 
I will answer here both on this issue and your other issue #54. 
The unfold macro is quite old - it predates the sheetmetal workbench. It was created by ulrich1a, and I later added it into the workbench. As a separate macro, it does not rely on any of the sheetmetal object history or properties and has it own, single k factor usage. 
The K factor usage in this macro was indeed different then the one in the wikipedia (by a factor of 2) so I modified his code to conform to the wikipedia.
About a year ago, another programmer joined the workbench, jaisekjames. He added a feature for that lets you set an individual K factor to each fold. To use it you need to set the unfold property of the fold to true, and this fold will be straighten. 
Of course, to unfold the entire model, you will need to set the unfold property one by one for all folds. 
Then someone found out that doing that (setting the unfold property of all folds to true) results in an object different then the one generated using Ulrich's macro. Turns out jaisekjames used the same k-factor calculations as Ulrich's original.
After contacting Ulrich using the forum, I was informed there are 2 standards for K factor (european and american, dont remember which is which) So I decided to revert the K factor to Ulrich's original, both because it is his code, and because it is compatible to jaisekjames's code.
You might ask why do we need 2 unfold systems? well, jaisekjames's is nice because you can unfold individual folds with individual k-factor. And Ulrich's is nice because it can unfold ANY model, not just one created by the sheetmetal workbench (I.E imported STEP model).
Regarding the formula, try asking ulrich on the forum (try finding the unfold macro thread)

shai

ceremcem, 2019-04-21:
So this issue has some historical reasons. 

There are 2 problems with the "2 different standarts" case: First is that if there are two standards because of two distinct definitions, they need two distinct namings, like `kilogram` and `pound`. They can't share the same `K-factor` naming because K-factor has no additional units. Second thing is that I could find no resource on the internet that indicates any other definition than the Wikipedia's. My best guess is that the `K-factor` which is the twice of the other *might be* actually the `Y-factor`, which is 1.57 times the `K-factor` and has a value of `0.5` by default (see [this](http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/sheetmetal/sheetmetaldesign/Set_Up/About_Y_factor_and_K_factor.html)):

> K factor is a ratio between the distance from the neutral bend line to the inside bend radius and the material thickness. K factor uses the formula K factor = δ/T. Y factor uses the formula Y factor = K factor * (Π/2). The default value for Y factor is 0.50.

**My proposal** would be that we should revert the definition of `K-factor` to the one we could find on the net. This will also allow new users to adapt quickly. (@jaisekjames) 

Regarding the formula: I've removed the formula from SheetMetal source code because a formula has nothing to do with the unfold operations. A formula can be used to determine the K-factor, but it should be a distinct facility (like a separate calculator) in this case.

IMO, K-factor is a crucial parameter of this application and the definition shouldn't be changed without a proper notification.

ulrich1a, 2019-04-21:
When I programmed the unfold macro, I searched for an "official" formula for the bend allowance. I found the formula based on a German standard: https://www.techstreet.com/standards/din-6935?product_id=1811065
It is explained in the wikipedia, but it is unfortunately in German language only:  https://de.wikipedia.org/wiki/Biegeverk%C3%BCrzung
The difference to ANSI is the definition of the neutral line. ANSI uses an index of 0.5 whereas the DIN-Standard uses an index of 1.0 for the neutral line. All what is needed is an explanation of the underlaying concept in some kind of help info. I was myself puzzled by two different formulas. I have chosen the DIN-formula, because it seems to be backed up by an industrial standard, though the ANSI-version seems to be mathematically more logical.

Ulrich

ceremcem, 2019-04-21:
@ulrich1a Thanks for the clarification. [Here](https://support.ptc.com/help/creo/ced_modeling/r20.2.0.0/en/index.html#page/ced_modeling%2FOSDM_Modules%2FSheetMetal_adm_allow.html%23wwID0EDVS3B) is another explanation I found after your explanation. 

By taking the comparison in search results into account, I think we should 
* stick to ANSI definition as it's easy to understand and has a simpler definition
* explicitly indicate that the K-factor is interpreted according to the ANSI standards
* Add a read-only input that indicates the calculated K-factor (x2) according to DIN standarts

What do you think?

closed 2019-04-24 by shaise