shaise/FreeCAD_SheetMetal #72: How to update Unfold Options menu via UnfoldOptions.ui?

ceremcem, 2019-04-24:
It turns out the `UnfoldOptions.ui` is not interpreted directly. AFAIU `pyuic4` is used to generate Python codes located in `SheetMetalUnfolder.py/SMUnfoldTaskPanel`. However, there are lots of code seems to be diverged from the output of `UnfoldOptions.ui`. 

What is the correct way to edit the UI?

shaise, 2019-04-24:
You are correct, this is my mess. I don't remember why I could not use the .ui directly, but this is the result. when I want to change it, I use pyuic4, then copy/paste over what the current lines are: 2162-2251

ceremcem, 2019-04-24:
I got it, it's OK. 

Regarding to using the `.ui` directly; I was using PySide a few years ago and [here](https://github.com/ceremcem/ggj-enzyme-game-backend/blob/da1b29338504fbfbbc6a5f25fe8d5eb01974b55c/display.py#L11) is the relevant part from one of my sample projects. The `Qt` wrapper is located [here](https://github.com/aktos-io/aktos-dcs-lib/blob/master/Qt/__init__.py). We may attempt to refactor the code to use `.ui` directly all in good time.

closed 2019-04-24 by ceremcem