shaise/FreeCAD_SheetMetal #75: missing default selection on Unfolding

easyw, 2019-04-26:
It would be useful to have defaults pre-checked
i.e. 'manual k-factor' checked and 'ANSI' or 'DIN' selected 
This would be best if it could be saved as in line 2148 as parameters
https://github.com/shaise/FreeCAD_SheetMetal/blob/master/SheetMetalUnfolder.py#L2148

[image]

ceremcem, 2019-04-26:
Hi @easyw, 

ANSI/DIN selection is left unselected for first unfold job on purpose because we need to be sure that user does know what he/she is doing. 

This is a crucial parameter as you would agree and applying a default selection might lead current users who expects that parameter to be in DIN standard to a terribly massed up situation in a production environment. 

As you might already notice, manual K-factor selection is required on every unfold job where K-factor standard is required only for the first time in a FreeCAD runtime.

easyw, 2019-04-26:
@ceremcem
I can understand your point of view, but if a user is going for a massive production without kmowing it or even doing some prototypes, then he,/her made a wrong job choice.
BTW those parameters choice are lost when you close FC.
In my previous post I suggested to save/load the preferences as in the linked code.

ceremcem, 2019-04-26:
> BTW those parameters choice are lost when you close FC. In my previous post I suggested to save/load the preferences as in the linked code.

That was also by intention. However, your point is also reasonable. I'll make a change to store the selection.

easyw, 2019-04-26:
@ceremcem 
thanks :smile:

shaise, 2019-04-26:
I also agree with @easyw , once you decide what selection to make, it should be saved.

ceremcem, 2019-04-27:
K-factor standard selection is now stored permanently. 

However, note that the stored K-factor standard is only regarded while providing a manual K-factor. If an unfold job uses Material Definition Sheet, the stored selection is simply ignored because a project might contain a third party library that depends on its different K-factor standard, so overall project might contain different K-factor standards.

easyw, 2019-04-27:
> However, note that the stored K-factor standard is only regarded while providing a manual K-factor. If an unfold job uses Material Definition Sheet, the stored selection is simply ignored

@ceremcem this is fine, and in that case (manual k_factor deselected) you may disable ANSI,DIN option selections (keeping just the saved value) ...
Or better, you may add a switch option between "manual k-factor" and "Material Definition Sheet" with the above behavior.

closed 2019-04-27 by shaise

easyw, 2019-04-27:
@ceremcem 
an other nicer option would be to have a field for the Material Definition Sheet Name in the dialog, that will be saved in the FC cfg. This would allow to avoid the need to rename the object to unfold accordingly to the sheet name.

shaise, 2019-04-27:
> 
> 
> @ceremcem
> an other nicer option would be to have a field for the Material Definition Sheet Name in the dialog, that will be saved in the FC cfg. This would allow to avoid the need to rename the object to unfold accordingly to the sheet name.

+1

ceremcem, 2019-04-27:
>  in that case (manual k_factor deselected) you may disable ANSI,DIN option selections

Added. 

> an other nicer option would be to have a field for the Material Definition Sheet Name in the dialog, that will be saved in the FC cfg.

Unfortunately this is not applicable. Material definitions are per folded object. This means that a project might contain multiple objects to unfold (say 10) and each one may use separate material definitions. A far better control of material definition sheets is already planned  but depends on the "Propagated Properties" feature: https://forum.freecadweb.org/viewtopic.php?f=3&t=35889#p303418 

It's currently unknown if the Propagated Properties will be available only in LinkStage3 or in FC main branch as well.

easyw, 2019-04-27:
@shaise it could be possible to add a text field to the feature python object editable with the sheet name ... would it be a viable/useful option?

shaise, 2019-04-27:
> 
> 
> @shaise could it be possible to add a text field to the feature python object editable with the sheet name ... would it be a viable/useful option?

It can be good if it is a property of the Body, rather then of the inner solid. (say the "Body Material"). This way you dont have to re-enter this text field every time you add more features to the solid.

PR #76 (2019-04-27): K-factor standard selection is persistent across FreeCAD runs
resolves #75
commit 94e0603a (2019-04-27): K-factor standard selection is persistent across FreeCAD runs

resolves #75