shaise/FreeCAD_SheetMetal #85: Need to adjust length of both sides independently

briankurzhal, 2019-07-04:
Need to breakout length into length1 and length2 to independently adjust each side of bends and extensions in order to create quadrilaterals.

shaise, 2019-07-06:
when different length needed, just make separate bend for each side

briankurzhal, 2019-07-06:
The same bend requires different side lengths.

shaise, 2019-07-06:
Sorry, but I do not understand what is the meaning of different lengths for a single bend.

briankurzhal, 2019-07-06:
Good question, and thank you for the opportunity to clarify.  It might be helpful if we first consider the Bend to be comprised of 2 parts, (radius and length).  Now, imagine the top view of a Bend with 180-degree angle and a 100mm length and 25mm radius.  Let us orient this Bend so that the radius is on the bottom and the length on top.  Then beginning at the bottom edge of the radius, and proceeding clockwise, consider the perimeter of the Bend being comprised of the following 6 contiguous segments:  1) base edge of radius;  2) left edge of radius;  3) left edge of length (100mm);  4) top edge of length;  5) right edge of length (100mm);  6) right edge of radius.  If the left and right edges of the length were permitted to be adjusted independently, then the top edge of the length could break free from its parallel relationship with its opposite bottom edge.  For example, if we could adjust the left edge of the length to be 110mm, and the right edge of the length to be 90mm, then we could create a convex quadrilateral.  We could apply the same scenario to the Extend.  If the left and right edges of an Extend were permitted to be adjusted independently, then the top edge could break free from its parallel relationship with its opposite bottom edge, and we could create a convex quadrilateral.

shaise, 2019-07-09:
Thanks for the clarification I now understand what you mean. 
Unfortunately the sheet metal code is heavily based on the fact its a metal sheet with constant width and perpendicular edges, so tapered edges and non concentric radiuses are not supported.

luzpaz, 2021-07-03:
@shaise should we rename this ticket then to `FR: support for tapered edges and non concentric radiuses` ?

shaise, 2021-07-04:
since the result model can not be flatten into a sheet, I don't think it should be sheet metal feature

luzpaz, 2021-07-04:
Where should this feature be relevant then?

shaise, 2021-07-04:
I can't say without understanding the use of it. you can reach the exact same result by adding a chamfer to the edge of a bend.