shaise/FreeCAD_SheetMetal #89: Bend gap twice as big as inserted value

ludwig1992, 2019-07-10:
Hi!

First of all, thank you for your great work on the sheetmetal workbench. I worked a lot with it in the past and it usually worked great. 

I use FreeCAD 0.18 stable with Debian on my computer at work.
Recently I noticed, that the gaps at sheet metal bends appear twice as big as i intended and I can reproduce it. To verify if the failure is not in my config, I renamed my .FreeCAD folder, to get a new FreeCAD without my old config files and installed a new sheetmetal workbench over the addon manager. Then I made a simple new model to verify if the bug still appears. And it did. My coworker has the same issue with the sheetmetal workbench on his FreeCAD. So I think that might be a bug. 

Please find a sample file and a screenshot attached. 
In this sample I intended a 5mm gap but I get 10mm. 

[bend_gap_clusterfuck_3.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/3373912/bend_gap_clusterfuck_3.zip)
[image]

I would be glad, if you could have a look on it to verify if its an actually bug.

Best regards,
Ludwig

shaise, 2019-07-12:
Hi Ludwig,

You are right. It a bug.
@jaisekjames Can you please check this issue? It looks like the miter function returns gaps twice as big, causing this bug.

shai

shaise, 2019-07-16:
Hi Ludwig,

It should be fixed now. can you check?

ludwig1992, 2019-07-16:
Hi shai,

it works perfect now! Thanks a lot!

closed 2019-07-16 by ludwig1992