shaise/FreeCAD_SheetMetal #93: Bending across sketch created axis not working

jaessy77, 2019-07-25:
Hi,
first: a really great workbench.

Im struggeling with the bending feature along a axis (which would be really helpful when folding rigid-flex PCB models).

Even a simple example wich a padded sketch fails at all.

SheetMetal is installed via Addon Manager and shows "Sheet Metal V0.2.31"

I select the face grom Pad_Plate and then the Sketch_Bendline and hit the button.
It ends up in a fold with exclamation mark and a non-folded result.

In Traceback I can see:
Traceback (most recent call last):
  File "C:\Users\mja\AppData\Roaming\FreeCAD\Mod\sheetmetal\SheetMetalFoldCmd.py", line 203, in execute
    invertbend = fp.invertbend, selFaceNames = fp.baseObject[1], MainObject = fp.baseObject[0])
  File "C:\Users\mja\AppData\Roaming\FreeCAD\Mod\sheetmetal\SheetMetalFoldCmd.py", line 104, in smFold
    edge = Axis.Edges[0]
<class 'IndexError'>: list index out of range

Do have the same problem ?

[image]

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16117 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: dbb4cc6415bac848a294f03b80f65e888d531742
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

jaisekjames, 2019-07-26:
shaise, Could you pls change line 90 & 92[instead of division to multiply] in SheetMetalFoldCmd.py to

offset = thk * kfactor
offset = thk - ( thk * kfactor )

to fix this bug

shaise, 2019-07-27:
done. thank you.

jaessy77, 2019-07-29:
Hi @jaisekjames  and @shaise ,

thank you very muich for the bug fix. Now it works propper.

You both did a great job !

Kind regards
Marco

[image]

closed 2019-07-29 by jaessy77