shaise/FreeCAD_SheetMetal #95: Auto-unfold STEP files

ahermle93, 2019-09-23:
We're currently using the Sheet Metal addon to unfold STEP files for calculation/production purposes.
Due to the manual effort we would like to automate the unfold process with a simple python script/tool using the FreeCAD Sheet Metal addon.
The user should simply drag & drop the STEP Files and receive the unfolded e.g. .DXF-Files or .PDF.
Is this possible with the Sheet Metal addon or is there such a tool already?

shaise, 2019-09-23:
try asking at the free cad forum: forum.freecadweb.org

luzpaz, 2020-02-23:
@ahermle93 did you post a thread? can you x-post it back here so we can track it ?

luzpaz, 2020-03-01:
I create the thread at https://forum.freecadweb.org/viewtopic.php?f=8&t=43796

Spoutnik97, 2023-02-13:
Me too ! Did you find a solution for that @ahermle93 ?
I thought about choosing automatically the biggest face instead of asking user to choose it

zuchtzucht, 2023-03-17:
Any update on this ?

Spoutnik97, 2023-12-04:
I will work on it in the next weeks / month. I'll keep you updated ;)