shaise/FreeCAD_SheetMetal #96: Detect features like welded screws

luzpaz, 2019-10-14:
One of the todos in `SheetMetalUnfolder.py`

luzpaz, 2021-03-03:
Hi @jaisekjames  
is this on your radar?

luzpaz, 2021-04-27:
@jaisekjames can you weigh-in?

jaisekjames, 2021-04-29:
No, I am not working this issue. I don't much Idea on unfolder coding.

Just an Idea.

We can select face that having this kind of deformation that cannot be unfolded. 

In Unfolder script, we can skip related faces that are inside that face so that same shape inside face is used.
[image]

In above image, If we can input skip face as top face & all deformed shape are inside that face. If we skip face without unfolding Unfolder will not fail & we will correct shape.

Only thing, we have to manually select faces required to skip.

luzpaz, 2022-08-17:
Anyone interested in tackling this ? :point_up:  @Syres916 ?