shaise/FreeCAD_SheetMetal #98: Add missing error codes to SheetMetalUnfolder.py

luzpaz, 2019-10-14:
There are certain places in the `SheetMetalUnfolder.py` that are missing error codes. 

* `# FIXME, need proper failure mode.`
* `print('FIXME! make errorcondition')`
* ```
  # - Put error numbers into the text
  # - Put user help into more texts
  ```

luzpaz, 2021-04-08:
Looks like @matthewdickinson's PR #10 was trying to address this

luzpaz, 2021-07-03:
@shaise was this actually realized?

shaise, 2021-07-04:
No, the PR was conflicting

luzpaz, 2022-11-09:
Updated ticket with hyperlinked references in to the code

alexneufeld, 2025-01-31:
@shaise This is probably not needed anymore, as we'll be using the new unfolder for the most part going forward

closed 2025-01-31 by luzpaz

shaise, 2025-02-02:
@luzpaz , Yes, with the new unfolder it is obsolete.  Thanks for helping me tidy things up!