shaise/FreeCAD_SheetMetal #107: Fold tool freezes the whole PC

ifohancroft, 2020-01-17:
I have found out about the Sheet Metal workbench yesterday and it seems it would be best suited for what I am trying to do.

However, it is unusable to me. I try to create a bend and it takes so much RAM and CPU it freezes my whole PC. I have tried it on a couple of different versions of FreeCAD (Development AppImage of version 19, stable AppImage of version 18, installed FreeCAD 18 from AUR) with the workbench always installed from the Addon Manager.

Doing this on Manjaro Linux.
The hardware is: CPU - Intel Core i7-6700k, 16GB Ram DDR4, Nvidia GeForce GTX1060 6GB

Is there currently a known issue with the workbench?

shaise, 2020-01-17:
Hi,

This WB is pretty simple and light, and as far as I know does not have any issues even on slow computers. Can you attach your file (zipped) with the problem and explain exactly what you are trying to do so we can reproduce this problem?

shai

ifohancroft, 2020-01-17:
Hi,

Thanks for the fast answer.
It may turn out that the problem is on my end or with FreeCAD as I have since discovered that I also have problem creating ball, conus and torus solids in the Part workbench (although without such detremental effects as my whole PC freezing).

Here is the exact FreeCAD detail version and stuff:
OS: Manjaro Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16146 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.4)
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

Here is the file:
[test.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/4079294/test.zip)

I am trying to do the following: I click on one of the side faces and then I click the bend button (the one whose tip says: Extends a wall from a side face of metal sheet. and that's it, after a couple of minutes, my ram is full and all my cores are used at 100% and my pc freezes. I left it overnight the other night and at the morning FreeCAD had crashed or something as it wasn't open anymore and my PC was responding again. I should have started it via the terminal so I can see if there was any error or some other output.

P.S. This is probably not related to the issue but is the workbench loaded during all times? I mean when starting FreeCAD from the terminal it right away says Sheet Metal workbench loaded, while it doesn't say so for the rest of the workbenches.

ifohancroft, 2020-01-18:
Sorry to waste your time. Apparently it was FreeCAD. Something in my config messed it up and it was causing the Fold tool in the Sheet Metal workbench and the Create Cylinder, Conus, Ball and Torus Solids in the Part Workbench to either freeze my pc or crash FreeCAD. By removing the configs, it's working flawlessly!

shaise, 2020-01-18:
Glad you figured it out!

luzpaz, 2020-02-23:
Please close ticket

ifohancroft, 2020-02-26:
Sorry, I forgot I can close it myself.

closed 2020-02-26 by ifohancroft