shaise/FreeCAD_SheetMetal #108: Unit tests to catch regressions introduced in development (either in master or in this repo)

luzpaz, 2020-01-19:
Can we discuss a way to run some unittests that catch possible regressions in this awesome addon. Case in point: https://forum.freecadweb.org/viewtopic.php?f=3&t=11303&p=361751#p361751  

I guess I should open a ticket in FreeCAD-addons requesting a tutorial on how to write a unit test and ask Addon authors/FC-community to do so for addons.

shaise, 2020-01-19:
addon unit test tutorial: +1

ceremcem, 2020-06-09:
Absolutely necessary. #122 wouldn't exist if there were appropriate unit tests. 

Moreover, unit tests would greatly help cleaning up the existing code. I would terrify if I start some cleanup because there is always a possibility to mess up the serious calculations.

ceremcem, 2020-12-09:
Ping @luzpaz 

Is there any clear instructions about writing unit tests?

luzpaz, 2020-12-09:
@ceremcem not that I know of off hand, sorry. Hope someone wrote some documentation for it. But if not, then it would be old-fashion look at the pre-exisiting unit-tests in the core workbenches for orientation + asking on the forum

ceremcem, 2020-12-14:
I would like to start with defining how it should be (according to me, of course): 

1. Start recording a macro.
2. Create a part from scratch or load it from a file. 
3. Perform relevant operations.
4. Stop the macro recording.
5. Create a Spreadsheet named `Tests` that will hold the expected values (lengths, angles etc.). It should contain "Description" (optional), "Expected" and "Observed" columns. `Expected` column should be hard coded and `Observed` column will be calculated by a formula.
6. Use another macro, named "Test.FCMacro", which will simply "Compare the `Expected` and `Observed` column values in the `Tests` Spreadsheet. 

This is how I expect to create a test for any workbench.

In order to execute tests: 
1. Execute FreeCAD by passing the filename, the relevant macro and the `Test.FCMacro` as arguments. I expect FreeCAD to load the file, execute first macro and then execute `Test.FCMacro`, return the output of `Test.FCMacro`. 

How does it sound?