shaise/FreeCAD_SheetMetal #121: [Unfold] Option to not add msd tag to label of original object

eht8, 2020-04-13:
I would like an option to prevent the unfold operation from changing the label of the original object. Perhaps a check mark in the unfold menu.

**Rationale**
When a bended piece of sheet metal is unfolded the tool adds a tag for the material definition sheet to the end of the object label. For instance "Body001" gets changed to "Body001_material_0.40ansi".

In some of my projects I spend a considerable amount of time at labelling parts and bodies. I don't like a tool altering those labels, especially since I don't (yet?) make use of material definition sheets anyway.

I appreciate other users may find it useful, so perhaps a check mark could be added to the unfold menu to give users the option to turn off this behaviour.

shaise, 2020-04-14:
You are absolutely right! 
Changing label names by tools is a bad practice, as I see it. This is a contributed code and it slipped my attention. Any way, as a quick fix, I changed the code such that if you do not use mds, no label changes will occur. Please test and let me know.

shai

eht8, 2020-04-16:
Did a quick test and this seems to work as intended.
Thanks for such a quick fix.

closed 2020-04-16 by eht8