shaise/FreeCAD_SheetMetal #125: To use at work

aspect-ex, 2020-05-05:
I work in the sheet metal business in practice.
In my company, we use a dedicated 2.5 DCAD.
In the case of SheetMetal Workbenche, the four sides of the rectangle are bent with MaterialInside and Radius 1.0 mm.
There will be a gap for Rasius on the side that bends later.
My ideal is to create in 3D with MaterialInside Rasius 0mm. On unfold, in Radius2.7 kfactor0.5
Is it possible to do this with the expansion specification?
I'm not very good at English, so I'm sorry if my writing is difficult to read.

shaise, 2020-05-05:
when you go to the bend properties (on the left), you can change the bend radius.

aspect-ex, 2020-05-05:
Thank you very much for your response.
Certainly, I can change in value to the properties of the data tab.
However, once I have completed the object by creating a 4-sided Bend with 0 mm MaterialInside radius, I can use
Then, if I change the radius to 2.0mm, I get an error on the last edge I created.
Unfold also fails and shows the following error.

Unfold is failing.
Please try to select a different face to unfold your object

shaise, 2020-05-05:
It looks like an operational issue. Please post your question in freecad forum: https://forum.freecadweb.org/ an you can also upload your file there so we can inspect it.

aspect-ex, 2020-05-05:
All right.
Thank you.

luzpaz, 2020-06-01:
Can this ticket be closed?

luzpaz, 2020-07-25:
@aspect-ex have you posted to the forum ? if so, please post the link here

luzpaz, 2020-08-19:
@aspect-ex ping

luzpaz, 2021-02-11:
Suggest this ticket be closed, as no response from OP. cc @shaise

shaise, 2021-02-11:
@luzpaz . Thanks. Closing due to inactivity.

closed 2021-02-11 by shaise