shaise/FreeCAD_SheetMetal #128: Error while trying to Extend a face along norma

Flopezconde, 2020-05-31:
Traceback (most recent call last):
  File "C:\Users\flope\AppData\Roaming\FreeCAD\Mod\sheetmetal\SheetMetalCmd.py", line 1124, in execute
    s = smExtrude(extLength = fp.length.Value, selFaceNames = face, selObject = Main_Object)
  File "C:\Users\flope\AppData\Roaming\FreeCAD\Mod\sheetmetal\SheetMetalCmd.py", line 1089, in smExtrude
    V_extDir = selFace.normalAt( 0,0 )
<class 'TypeError'>: function takes exactly 1 argument (2 given)

I am a new use of FreeCAD and your workbench.  This is the software version in use:
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

shaise, 2020-05-31:
I do not see such issue. Can you explain step by step what you did, and perhaps attach the problematic file (zipped)

Flopezconde, 2020-05-31:
Thanks for your fast response.  The attached .zip file contains the FreeCAD file I was working on and the pictures are the lesson I was following from FREECAD 0.18 BASICS TUTORIAL book.  The pictures details the steps the I followed when encounter the error.
[SheerMetalModeling Files.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/4708432/SheerMetalModeling.Files.zip)

shaise, 2020-06-01:
OK. Firat I found some errors in your model (flanges are intersecting) so I redo the tutorial. Regarding extends, there is an error in the book. you do not extend an edge. you need to select the Face you want to extend, not the edge. and a single one each time. I did not see the next page, so I do not know what is the expected result, but attached the fixed file.
[Tutorial 7_1f.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/4709978/Tutorial.7_1f.zip)

Flopezconde, 2020-06-01:
Thanks very much.  I didn't know if I was doing something wrong.  I am in the process to inform the book publisher of your findings.  The attached .zip is the remaining pages of the lesson in case you are interested looking at the final product.

Thanks again!
[SheerMetalModeling Pages.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/4712443/SheerMetalModeling.Pages.zip)

Flopezconde, 2020-06-01:
When I unzip your fixed file it only contained the FCstd1 and I don't know how to use it.

shaise, 2020-06-01:
Sorry: here is the correct one:
[Tutorial 7_1f.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/4712576/Tutorial.7_1f.zip)

luzpaz, 2021-04-08:
Can be closed

closed 2021-04-08 by shaise