shaise/FreeCAD_SheetMetal #129: SheetMetal isn't represented in the FreeCAD dropdown when Workbench is loaded.

luzpaz, 2020-06-01:
Most workbenches, once they are loaded, will insert a dropdown in to the FC main dropdown menu between **Macro** and *Windows* of their own workbench commands. SheetMetal doesn't do this. 
Sketcher adds **Sketch**  
TechDraw adds **TechDraw**  etc... 
Lets consider adding this for folks who want to access the SheetMetal commands via the dropdown

shaise, 2020-06-01:
Is there any doc or example how to do it?

luzpaz, 2020-06-01:
CC @carlopav  
With all you work on Draft WB, can you illuminate where the code for adding commands to the FC dropdown menu is (as an exmaple) ?

carlopav, 2020-06-01:
it's very easy, just try to decomment this line!
https://github.com/shaise/FreeCAD_SheetMetal/blob/d6a33ed3a40ca5f4a71a70093eac428b267da548/InitGui.py#L58

carlopav, 2020-06-01:
@luzpaz I still have the doubt if this was what you were asking :)

shaise, 2020-06-01:
Well, this was rather easy... 
And yes, this is what @luzpaz meant. Thanks @carlopav!

carlopav, 2020-06-01:
:D

shaise, 2020-06-01:
It was slightly more complicated on the Fasteners WB, but added it there as well.

luzpaz, 2020-06-01:
Perfect. Thank you both! :smile:  
Closing ticket

closed 2020-06-01 by luzpaz