shaise/FreeCAD_SheetMetal #135: Include the "Fold a wall of MS" and "Create bend on solids" tutorials

ceremcem, 2020-07-09:
"Fold a wall of Metal Sheet" and "Create bend on solids" tutorials should be included or linked in the README of this repo. As an example, I'm struggling to find my way out to use them. 

* Question/answer form of "Fold a wall of MS": https://forum.freecadweb.org/viewtopic.php?t=45519#p389955 (does not work for me)

shaise, 2020-07-09:
Agree. @jaisekjames , do you have any tutorials on these functions?

jaisekjames, 2020-12-23:
First draw a line in sketch [ length more than face ] on face need to bend.
[image]

Then select face & sketch together, click on folding icon [ due to some issue, green face selection is not capturing in my screenshot.]
[image]

It will be folded.
[image]

In case in two line in sketch[co-linear], it will create fold only where line is resent
[image]

Folded view
[image]

RRhino8687, 2026-10-16:
This issue was made before the relevant wiki pages were made ([SheetMetal_AddFoldWall](https://wiki.freecad.org/SheetMetal_AddFoldWall), and [SheetMetal_AddBend](https://wiki.freecad.org/SheetMetal_AddBend)). It seems like the documentation is reasonably and the wiki is linked from the Readme.

Can this be closed, or would you like direct links added to the Readme?

shaise, 2026-10-17:
Agree, there is no sense having an individual link for each sheet metal command. Closing.

closed 2026-10-17 by shaise