shaise/FreeCAD_SheetMetal #138: Not loading on 0.18

dham-eng, 2020-07-13:
Hello,

I can't load this workbench on 0.18.4 anymore:
```
name 'os' is not defined
Traceback (most recent call last):
  File "<string>", line 66, in Initialize
```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Portuguese/Brazil (pt_BR)

With CAELinux 2018 (Xubuntu 16.04 LTS) the result is the same, with Freecad 0.18.4 Py3/QT5 and Py2/QT4.

closed 2020-07-20 by dham-eng

ASzc, 2020-09-11:
I'm seeing this error with the latest stable appimage (FreeCAD_0.18-16146-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage), and the latest version of this module (V0.2.43 / 206bd12)

This is the output if I run freecad from a terminal:

```
alex@elendil ~> freecad
/tmp/.mount_freecaFG70kU
FreeCAD 0.18, Libs: 0.18R16146 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
name 'os' is not defined
Traceback (most recent call last):
  File "<string>", line 67, in Initialize
```

The traceback appears when I try to switch into the Sheet Metal workbench via the GUI, and the workbench doesn't load.

ASzc, 2020-09-11:
Filed a fix for what I'm seeing in PR #142

PR #142 (2020-09-12): Import os.path in InitGui.py Initialize function
When using the workbench on FreeCAD 0.18.4 Linux AppImage, this error
prevents it from loading:

```
name 'os' is not defined
Traceback (most recent call last):
  File "<string>", line 67, in Initialize
```

I have no idea why the global import added in commit 206bd12 doesn't
work, but I tried several different things, and this change fixed the
error for me.

This (re) fixes #138
commit 534b28b6 (2020-09-12): Import os.path in InitGui.py Initialize function

When using the workbench on FreeCAD 0.18.4 Linux AppImage, this error
prevents it from loading:

```
name 'os' is not defined
Traceback (most recent call last):
  File "<string>", line 67, in Initialize
```

I have no idea why the global import added in commit 206bd12 doesn't
work, but I tried several different things, and this change fixed the
error for me.

This (re) fixes #138