shaise/FreeCAD_SheetMetal #140: Unfold exception occur

nik570, 2020-08-12:
Unfold fail with exception with exception below

, have a file but are not able to submit. Removing the last added bend and it will be OK but have not had time to look into code what might be wrong.

`16:18:06  exception at line 2876('list indices must be integers or slices, not NoneType',)
16:18:06  ('list indices must be integers or slices, not NoneType',)
16:18:06  Traceback (most recent call last):
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 2876, in accept
    s, foldComp, norm, thename, err_cd, fSel, obN  = getUnfold(k_factor_lookup)
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 2174, in getUnfold
    TheTree.Bend_analysis(f_number, None) # traverses the shape and builds the tree-structure
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 1329, in Bend_analysis
    self.Bend_analysis(child_info[0], t_node, child_info[1])
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 1329, in Bend_analysis
    self.Bend_analysis(child_info[0], t_node, child_info[1])
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 1329, in Bend_analysis
    self.Bend_analysis(child_info[0], t_node, child_info[1])
16:18:06    [Previous line repeated 3 more times]
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 1320, in Bend_analysis
    t_node = self.make_new_face_node(face_idx, parent_node, parent_edge, wires_edge_lists)
16:18:06    File "/home/karlnick/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 1103, in make_new_face_node
    oppVertIndexes = range(len(self.__Shape.Faces[newNode.c_face_idx].Vertexes))
16:18:06  TypeError: list indices must be integers or slices, not NoneType
16:18:06  `

lilaL, 2020-11-17:
[image]
There is an error in the whole thing I guess.. I mean how should the unfold looks like?

nik570, 2020-11-22:
Checking again it look OK here
[image]

lilaL, 2020-11-22:
Did you recompute the model?

luzpaz, 2020-12-19:
@nik570 bump

nik570, 2020-12-19:
git pull to the latest version. It did not unfold original file I sent but adding the last failing bend again on part an it unfold so something must have been fixed.

I could see there was changes to bot unfold and other files. Maybe I should press the close with comment button?

luzpaz, 2020-12-22:
> It did not unfold original file I sent  

Why?

luzpaz, 2021-03-03:
bump @nik570

luzpaz, 2021-04-08:
close ticket?

luzpaz, 2021-04-13:
@shaise probably can be closed

closed 2021-04-14 by shaise