shaise/FreeCAD_SheetMetal #143: Wrong sheet metal generation from "sheetmetal wall from sketch" feature.

dham-eng, 2020-09-24:
Hello,

I often find myself trying to heal these sheet metal features.

Sometimes it work as intended, sometimes, it does not.

Please note that I tested this command inside and outside PartDesign's Bodies, inside and outside Draft's layers, also.

[image]
[image]

Workbench is fully updated.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22411 (Git)
Build type: Release
Branch: master
Hash: d4519d5aec15bc08f029da5ed15410a494b57b2c
Python version: 3.8.5
Qt version: 5.12.6
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Portuguese/Brazil (pt_BR)

jaisekjames, 2020-12-01:
is this issue still there. pls send any problematic file or step to produce.

dham-eng, 2020-12-08:
Hello, jaisekjames,

sorry for the late response.

I tested the updated workbench on a new file and it worked fine.

On the file I showed initially when I try to recalculate it shows "name 'obj' is not defined".

When deleting and recreating the basebend on the same body the length got scaled by a factor of 1000 ( you need to use 1.200mm to get 1200mm of real extrusion). (File attached, you can try this

When creating a basebend on another body but with sketch being moved in space it gives the error "there are no suitable edges for chamfer or filet". On the other hand, moving a body with a internal basebend works fine.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23141 (Git)
Build type: Release
Branch: master
Hash: 0eba78d40096720f7a637a9a73c382a08b820bca
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Portuguese/Brazil (pt_BR)

[SheetMetal.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/5659691/SheetMetal.zip)

jaisekjames, 2020-12-23:
Fixed in above in 
https://github.com/shaise/FreeCAD_SheetMetal/pull/159

pls check it.

dham-eng, 2020-12-23:
Hello, jaisekjames,

the lenght issue is solved.

The only thing that still wrong it's the sketch's position bug, where you can not move it in x or y direction manually (it worked for me attaching the sketch to a shapebinder on a test instead of directly moving the sketch through it's properties/equations).

Steps to reproduce:
1 - Create a basebend
2 - Move placement/attachment coordinates of sketch (x or y).

jaisekjames, 2020-12-24:
Fixed in above commit. pls check it.

luzpaz, 2021-07-03:
@dham-eng ping