shaise/FreeCAD_SheetMetal #169: Corner Relief error

Softtechtb, 2020-12-30:
I get the following error when trying to make corner relief.

Traceback (most recent call last):
  File "C:\Users\Titus\AppData\Local\FreeCAD 0.18\Mod\SheetMetal\SheetMetalCornerReliefCmd.py", line 441, in execute
    MainObject = fp.baseObject[0])
  File "C:\Users\Titus\AppData\Local\FreeCAD 0.18\Mod\SheetMetal\SheetMetalCornerReliefCmd.py", line 320, in smCornerR
    facelist = resultSolid.ancestorsOfType(BendEdge, Part.Face)
<class 'UnboundLocalError'>: local variable 'BendEdge' referenced before assignment

jaisekjames, 2020-12-31:
pls attach a sample with problem for checking.

Softtechtb, 2020-12-31:
Here is the file I have a problem with.
[SheetMetalTest.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/5757522/SheetMetalTest.zip)

Windows 8.1
64 bit
FreeCad version 18
SheetMetal workbench version 2.44

jaisekjames, 2021-01-01:
Size 3mm too small to make a cut.

Adjust radius to 11-12mm to create relief

or Change circle to circle scaled & set above 1.0.

Softtechtb, 2021-01-01:
Ok Thanks that solved my problem!!

closed 2021-01-01 by Softtechtb